TimSoethout/goodwe-sems-home-assistant #1: Adhere to new HA naming, introduced in version 0.88

bouwew, 2019-02-23:
FYI, the sensors should now be in its own directory, with the name sensor.py

So, for the sems sensor:
../custom_components/sems/sensor.py

TimSoethout, 2019-02-26:
Thanks. I'll update it as soon as I'm updating HA myself.

Feel free to do the change in the meantime and make a PR. :)

JoLambinon, 2019-03-25:
Hello Everyone
I have been happily using the sems scraper created by Tim, but since the new upgrade for HA as mentioned by bouwew my sensor is not working anymore.  Will you please update GitHub so complete beginners like me can understand how to get it running again.

TimSoethout, 2019-04-14:
Should be fixed in https://github.com/TimSoethout/goodwe-sems-home-assistant/commit/c79a902910e0465a3b2ad753badfe71102c89cf5.

Please let me know if it is unclear.

closed 2019-04-14 by TimSoethout