TimSoethout/goodwe-sems-home-assistant #2: HA Naming

stringy92, 2019-04-19:
I am converting the custom component from Sensors -> Sems.py into Sems -> sensor.py, however when i do this, i get that Sensor_sems isnt a valid component. I am yet to upgrade to 91.4 from 90.1, as i wanted to make sure this worked before going ahead. Any help would be great!

TimSoethout, 2019-04-19:
I'm running version 0.91.3 and my files are in: `custom_components/sems/`
named sensor.py.
Seems to work fine. Maybe the older version doesn't like it?
I do get a warning when starting up about using an unsupported custom
component, but I think we can ignore that.

TimSoethout, 2019-04-19:
Do you see any specific error messages?

stringy92, 2019-04-25:
I reverted back to a previous setup and upgraded and it worked.

closed 2019-04-25 by stringy92