TimSoethout/goodwe-sems-home-assistant #3: HA 0.92 Manifest Creation

stringy92, 2019-04-25:
Just upgrading to 0.92 and for custom components you require a Manifest.json for each component - https://developers.home-assistant.io/docs/en/creating_integration_manifest.html

 I have created one which seems to work shown below;

`{
  "domain": "sems",
  "name": "Sems Scrape",
  "documentation": "https://github.com/TimSoethout/goodwe-sems-home-assistant",
  "dependencies": ["sensor"],
  "codeowners": ["@TimSoethout"],
  "requirements": []
}`

bouwew, 2019-04-26:
It works for me with the following contents:
```{
  "domain": "sems",
  "name": "Goodwe SEMS scraper",
  "documentation": "https://github.com/TimSoethout/goodwe-sems-home-assistant",
  "dependencies": [],
  "codeowners": [],
  "requirements": []
}
```

TimSoethout, 2019-05-27:
Fixed by #4.

closed 2019-05-27 by TimSoethout