TimSoethout/goodwe-sems-home-assistant #7: data is not updated

byjosemaria, 2019-07-21:
Hello, I have installed this component in Hassio and the data is not updated, only if I restart Hassio makes a new reading of the data. regards

TimSoethout, 2019-07-21:
Hi, that is odd. I do have sometimes that it looses connection to the SEMS portal and then a restart of home assistant makes is work again.

Do you mean you get a new single data point when restarting? I think that is just the last known value, and not really fresh data. 
Does your home assistant log say anything? You should be able to see that on the `http://hassio:8123/dev-info` page.

byjosemaria, 2019-07-21:
Hello
It was my mistake. It works perfectly. 
I was confused when I checked the time the sensor was always increasing, I thought it was reset every time I updated the data with the SEMS portal.
Thanks and regards

TimSoethout, 2019-07-21:
NP, glad it is working!

closed 2019-07-21 by TimSoethout