TimSoethout/goodwe-sems-home-assistant #8: Unable to fetch data from SEMS. list index out of range

kaizersoje, 2019-11-13:
I started getting this error "Unable to fetch data from SEMS. list index out of range" since yesterday. Any idea what to do to fix this?

vPotusFlotus, 2019-11-14:
Hi!

I'm experiencing the same issue. 
It appears to me that the HTML response is now different, the python script fails on the following line of code:

`filter1 = str(soup).split("var pw_info = ")[1]`

The HTML response does not contain any text matchin 'var pw_info =' anymore, hence we get this index out of range error. 

I noticed that the SEMS portal has an API! (finally!) My proposal would be to maybe use the API as SEMS integration for Home Assistant? 

Below you can find a short explanation:
1. Get the ID of your Solar Panels (Power Station) from the SEMS Portal. This ID is located in the URL once you access the SEMS Plant Status portal. If you look at the source of the page, you’ll also find it under powerstation_id or pw_id, but in the **URL** it’s super easy and straight forward.
https://www.semsportal.com/powerstation/powerstatussnmin/**12345678-1234-1234-1234-123456789012**
2. In your configuration.yaml  add a parameter under the SEMS platform lines called 'id' . The confiugration.yaml file for this component looks as follows:
```
  - platform: sems
    username: 'yourUsername@something.com'
    password: 'yourPasswordOrSecretsFileReference'
    id: 'yourPowerStationID e.g.: 12345678-1234-1234-1234-123456789012'
    scan_interval: 600
```

3. I modified the sems.py file to make the REST API calls now:
`"""
Support for power production statistics from Goodwe SEMS web portal.

For more details about this platform, please refer to the documentation at
https://github.com/TimSoethout/goodwe-sems-home-assistant
"""

import json
import logging

import requests
import voluptuous as vol

from homeassistant.components.sensor import PLATFORM_SCHEMA
from homeassistant.const import CONF_PASSWORD, CONF_USERNAME, CONF_ID
import homeassistant.helpers.config_validation as cv
from homeassistant.helpers.entity import Entity

# Validation of the user's configuration
PLATFORM_SCHEMA = PLATFORM_SCHEMA.extend({
    vol.Required(CONF_USERNAME): cv.string,
    vol.Required(CONF_PASSWORD): cv.string,
    vol.Required(CONF_ID): cv.string
})

_LOGGER = logging.getLogger(__name__)

_URL = 'https://www.semsportal.com/api/v1/Common/CrossLogin'
_PowerStationURL = 'https://www.semsportal.com//api/v1/PowerStation/GetMonitorDetailByPowerstationId'

def setup_platform(hass, config, add_devices, discovery_info=None):
    """Setup the GoodWe SEMS portal scraper platform."""
    # Add devices
    add_devices([SemsSensor("SEMS Portal", config)], True)

class SemsSensor(Entity):
    """Representation of the SEMS portal."""

    def __init__(self, name, config):
        """Initialize a SEMS sensor."""
        # self.rest = rest
        self._name = name
        self._config = config
        self._attributes = {}

    @property
    def name(self):
        """Return the name of the sensor."""
        return self._name

    @property
    def state(self):
        """Return the state of the device."""
        return self._attributes['status']

    @property
    def device_state_attributes(self):
        """Return the state attributes of the monitored installation."""
        return self._attributes

    def update(self):
        """Get the latest data from the SEMS API and updates the state."""
        _LOGGER.debug("update called.")
        try:
        # Get our Authentication Token from SEMS API
            _LOGGER.debug("SEMS - Getting API token")

            # Prepare Login Headers to retrieve Authentication Token
            login_headers = {
                'Content-Type': 'application/json',
                'Accept': 'application/json',
                'token': '{"version":"v2.1.0","client":"ios","language":"en"}',
            }

            # Prepare Login Data to retrieve Authentication Token
            login_data = '{"account":"'+self._config.get(CONF_USERNAME)+'","pwd":"'+self._config.get(CONF_PASSWORD)+'"}'

            # Make POST request to retrieve Authentication Token from SEMS API
            login_response = requests.post(_URL, headers=login_headers, data=login_data )

            # Process response as JSON
            jsonResponse = json.loads(login_response.text)

            # Get all the details from our response, needed to make the next POST request (the one that really fetches the data)
            requestTimestamp = jsonResponse["data"]["timestamp"]
            requestUID = jsonResponse["data"]["uid"]
            requestToken = jsonResponse["data"]["token"]

            _LOGGER.debug("SEMS - API Token recieved: "+ requestToken)
        # Get the status of our SEMS Power Station
            _LOGGER.debug("SEMS - Making Power Station Status API Call")

            # Prepare Power Station status Headers
            headers = {
                'Content-Type': 'application/json',
                'Accept': 'application/json',
                'token': '{"version":"v2.1.0","client":"ios","language":"en","timestamp":"'+str(requestTimestamp)+'","uid":"'+requestUID+'","token":"'+requestToken+'"}',
            }

            data = '{"powerStationId":"'+self._config.get(CONF_ID)+'"}'            

            response = requests.post(_PowerStationURL, headers=headers, data=data)

            # Process response as JSON
            jsonResponseFinal = json.loads(response.text)

            _LOGGER.debug("REST Response Recieved")

            for key, value in jsonResponseFinal["data"]["inverter"][0]["invert_full"].items():
                if(key is not None and value is not None):
                    self._attributes[key] = value
                    _LOGGER.debug("Updated attribute %s: %s", key, value)
        except Exception as exception:
            _LOGGER.error(
                "Unable to fetch data from SEMS. %s", exception)
`
This works for my Home Assistant instance, running 0.91.1. Feel free to test / modify it to your coding standards.

TimSoethout, 2019-11-14:
Not yet. Thanks for reporting.

I'll take a look when I'm back home from holidays next week.
Feel free to try to find the bug in the mean time and PR. They might have
changed the web portal.

vPotusFlotus, 2019-11-14:
Created PR. See: https://github.com/TimSoethout/goodwe-sems-home-assistant/pull/9 

Feel free to let me know what you think! :-) 

Have a nice day!

kaizersoje, 2019-11-14:
Thx @vPotusFlotus. It works for my Home Assistant instance, running 0.101.3

byjosemaria, 2019-11-14:
I have error to validate config.yaml:

"Invalid config for [sensor.sems]: [id] is an invalid option for [sensor.sems]. Check: sensor.sems->id."

Home Assistant 0.101.3
Regards

vPotusFlotus, 2019-11-15:
Great to hear it's working! :-) Enjoy the weekend!

TimSoethout, 2019-11-19:
#9 merged. Thanks a lot @vPotusFlotus!

closed 2019-11-19 by TimSoethout