TimSoethout/goodwe-sems-home-assistant #10: More than 1 inverter

jojoro1, 2019-11-24:
How to get data from second inverter?

TimSoethout, 2019-11-24:
Hi @jojoro1 , this is currently not possible, since you can only add one converter via configuration.

If we want to support this, I guess we should allow to configure multiple station id's and create separate sensors for them. Feel free to create a pull request for this.

TimSoethout, 2019-11-24:
https://github.com/TimSoethout/goodwe-sems-home-assistant/blob/master/custom_components/sems/sensor.py#L36 should be called multiple times with different station id configuration.

spaya1, 2020-05-06:
Hi Tim,

I use this plugin with one inverter and I love it, I’m in the process of extending the system with a second inverter and I saw that it’s not possible by default with this plugin. I was wondering if you ever got to update the code to add multiple inverters. I have no clue about coding so want to know if it would be possible to add the functionality. 

Appreciate if you can let me know so that when it’s not possible I look for a different brand of inverter that can be integrated in HA, so that I can run them both in HA as a single sensor.

Thx for the time and for keeping this alive.

Sergio

TimSoethout, 2020-05-07:
Hi Sergio, I did not add this to the code yet, simply because I do not need the feature myself at the moment. Of course anyone is free to add it, and I will gladly merge it. Adding multiple inverters is definitely useful. I might do it when I have some spare time, but don't count on it. Also I can't test myself if it works or not, since I do not have two inverters.

Ideally the configuration should not have to change for people just having one inverter: so simpel config for single device and some more involved for more: `devices: [ list of inverter configs ]`.
Although I'm not sure if this style is idiomatic style for home assistant component.

Also what do you think such a feature would look like from a user perspective?

@spaya1 , maybe for you a quick fix (hack) could be:
- make a copy of the `sems` directory in custom components, name it `sems2`
- in `sems2/sensor.py` change [line 37](https://github.com/TimSoethout/goodwe-sems-home-assistant/blame/master/custom_components/sems/sensor.py#L37) to use some other name: `add_devices([SemsSensor("SEMS Portal **2**", config)], True)`
- copy your configuration for `sems` to `sems2` and change the relevant fields.

I have not tested this, but it just might work.

spaya1, 2020-05-07:
Hi Tim,

Thanks for coming back so fast. I have tried the hack but it failed.

I have copied the sems folder to sems2 , so now in custom component I have both files.

I have duplicated config to look like this:

  - platform: sems
    username: user
    password: pass
    station_id : id1
    scan_interval: 60
 
  - platform: sems2
    username: user
    password: pass
    station_id : id1
    scan_interval: 60

I dont have yet the second inverter, so I have copied the same station id twice.

I get two errors:

**Error 1:**
Logger: homeassistant.components.sensor
Source: custom_components/sems2/sensor.py:57
Integration: Sensor (documentation, issues)
First occurred: 10:28:36 AM (1 occurrences)
Last logged: 10:28:36 AM

Error while setting up sems2 platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 186, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 295, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 445, in _async_add_entity
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 297, in async_write_ha_state
    self._async_write_ha_state()  # type: ignore
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 320, in _async_write_ha_state
    state = self.state
  File "/config/custom_components/sems2/sensor.py", line 57, in state
    return self._attributes['status']
KeyError: 'status'

**Error2:**
Logger: custom_components.sems2.sensor
Source: custom_components/sems2/sensor.py:118
Integration: sems2 (documentation)
First occurred: 10:28:36 AM (1 occurrences)
Last logged: 10:28:36 AM

Unable to fetch data from SEMS. 'NoneType' object is not subscriptable

TimSoethout, 2020-05-07:
Maybe you need to change https://github.com/TimSoethout/goodwe-sems-home-assistant/blob/master/custom_components/sems/manifest.json also to `sems2`?

spaya1, 2020-05-07:
Hi Tim, i did change it before restarting HA.
{
        "domain": "sems2",
        "name": "Goodwe SEMS API",
        "documentation": "https://github.com/TimSoethout/goodwe-sems-home-assis$
        "dependencies": [],
        "codeowners": ["@TimSoethout"],
        "requirements": []

TimSoethout, 2020-05-07:
I'm not sure what goed wrong here. You could try turning on debug logging and see if it gives us more information: https://www.home-assistant.io/integrations/logger/

spaya1, 2020-05-07:
Hi Tim,

It seems that sems2 is working now but sems is not :)

I added this:

logger:
  default: info
  logs:
    homeassistant.components.sems: debug

I am getting this new error:

Logger: custom_components.sems.sensor
Source: custom_components/sems/sensor.py:118
Integration: sems (documentation)
First occurred: 11:28:51 AM (1 occurrences)
Last logged: 11:28:51 AM

Unable to fetch data from SEMS. HTTPSConnectionPool(host='www.semsportal.com', port=443): Read timed out. (read timeout=30)

TimSoethout, 2020-05-07:
You can add `homeassistant.components.sems2: debug` as well for more info.

It seems that the api does not like 2 requests (for the same inverter) at the same time maybe?

spaya1, 2020-05-07:
Hi Tim,

I have added it, but it doesnt add any new information besides what I have added earlier:

Error 1:

Logger: homeassistant.components.sensor
Source: custom_components/sems2/sensor.py:57
Integration: Sensor (documentation, issues)
First occurred: 11:45:07 AM (1 occurrences)
Last logged: 11:45:07 AM

Error while setting up sems2 platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 186, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 295, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 445, in _async_add_entity
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 297, in async_write_ha_state
    self._async_write_ha_state()  # type: ignore
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 320, in _async_write_ha_state
    state = self.state
  File "/config/custom_components/sems2/sensor.py", line 57, in state
    return self._attributes['status']
KeyError: 'status'

Error 2:

Logger: custom_components.sems2.sensor
Source: custom_components/sems2/sensor.py:118
Integration: sems2 (documentation)
First occurred: 11:45:07 AM (1 occurrences)
Last logged: 11:45:07 AM

Unable to fetch data from SEMS. 'NoneType' object is not subscriptable

I tried to install the mqtt2sems plugin, it works but it makes my HA system crash, (had to restart my raspberry 3 times in the mean time, so I disabled it for now) so want to continue your route if possible.

TimSoethout, 2020-05-07:
Oke. Well it should be definitely possible, but ideally we have someone with 2 inverters already to try this out. Maybe the SEMS API system blocks simultaneous logins or requests. Or it is because you're using the same inverter id twice.

Also this hack is taking too much effort already compared to just implementing multiple inverters correctly. :P
If I have some free time tonight, I might start implementing it. :)

spaya1, 2020-05-07:
Well if its too much work i can go with Growatt for the same price, so I could interface both brands under one sensor, would prefer Goodwe as it has higher efficiency, but I also dont want to force it. In any case, they said it would take 4 weeks before they are able to install so there is time to continue playing :)

spaya1, 2020-05-26:
Hello Tim, hope you are fine. I got the second inverter installed :)

As expected it only reads one. I have setup both inverters under the same plant, and under the sems sensor in HA it only detects the first inverter (new one) and the old one has dissappeared.

In the sems application it does detect the increase in power as you can see below.

[image]

TimSoethout, 2020-05-26:
That makes sense, because we only add the values of the first inverter: `jsonResponseFinal["data"]["inverter"][0]["invert_full"].items()`

To show the other inverters, we would need to add the values of all inverters, by looping over `jsonResponseFinal["data"]["inverter"], which should be a list of inverters.
But how should we name the values?
- Do we expect multiple devices in HA, one for each inverter? 
- Or one global SEMS device (as it is now), with prefixed inverter ids or something?

The first seems more proper, but also harder to implement if we don't want to do multiple API request per plant.

spaya1, 2020-05-26:
For me the ideal case is when you have n sensors, so SEMS_Portal_n, that way you can monitor the inverters separately, and add them together under one sensor if you want.

I dont expect many people to have more than 2 inverters, I also think having two inverters is not common (in my case it was 350e vs 950e to go for 2 inverters vs 1 big one).

Option two is possible, you just need to call the right attribute under a new sensor and you are done right?

TimSoethout, 2020-05-28:
I'm building a new version in this branch: https://github.com/TimSoethout/goodwe-sems-home-assistant/tree/%2310-multiple-inverters%2Basync

It is not finished, but it should display multiple inverters. Their names are the inverters serial numbers for now, but I plan to change this.

Can you test if it indeed adds multiple devices for each of your inverters?

spaya1, 2020-05-29:
Hi Tim, good morning.

Thanks for the effort, I have loaded the branch but see the same sensors created with only the new inverter showing (same as before).

Do the sensors have a different name now? I have a GW2000-ns and GW2000-xs, when I search in the state tab for "2000" I only get returned GW2000-xs in the sems_portal sensor.

TimSoethout, 2020-05-29:
They should have diffeerent names. If you enable debug logging you should see something like this in the logs: `DEBUG (MainThread) [custom_components.sems.sensor] Found inverter attribute GoodweName X3000SSNXXXXX`

spaya1, 2020-05-29:
Hi Tim,

I must be doing wrong, I have set it up under sems3, and this is my config:

  - platform: sems3
    username: 
    password: 
    station_id:
    scan_interval: 60

logger:
  default: info
  logs:
    homeassistant.components.sems3: debug

I have extended the log and all I can see with sems is this:

2020-05-29 08:20:53 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for sems3 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.

2020-05-29 08:20:55 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.sems3

but beyond this, there is no more info

spaya1, 2020-05-29:
nvm my logger was wrongly configured...

2020-05-29 08:28:57 DEBUG (SyncWorker_6) [custom_components.sems3.sensor] REST Response Recieved
2020-05-29 08:28:57 DEBUG (SyncWorker_6) [custom_components.sems3.sensor] Updated attribute sn: ----
2020-05-29 08:28:57 DEBUG (SyncWorker_6) [custom_components.sems3.sensor] Updated attribute powerstation_id: ----
2020-05-29 08:28:57 DEBUG (SyncWorker_6) [custom_components.sems3.sensor] Updated attribute name: ----
2020-05-29 08:28:57 DEBUG (SyncWorker_6) [custom_components.sems3.sensor] Updated attribute model_type: GW2000-XS

This is the only one it finds

TimSoethout, 2020-05-29:
You should see something like this:
```
2020-05-28 22:23:38 DEBUG (MainThread) [custom_components.sems.sensor] REST Response Received
2020-05-28 22:23:38 DEBUG (MainThread) [custom_components.sems.sensor] Found inverter attribute GoodweName SNXXXXX
2020-05-28 22:23:38 DEBUG (MainThread) [custom_components.sems.sensor] Found inverter attribute GoodweName SNXXXXX (Second inverter)
2020-05-28 22:23:38 DEBUG (MainThread) [custom_components.sems.sensor] Finished fetching SEMS Portal data in 0.557 seconds
``` (maybe with `sems3` for you).

Are you sure you are running the new version? Those "update attribute XXXX" for each attribute should not be visible in the new version.

spaya1, 2020-05-29:
Hi Tim,

I am pretty sure I do, I have opened the file and compared with the branch code.

I have reset again and now get a new error:

Logger: homeassistant.util.async_
Source: util/async_.py:120
First occurred: 8:55:39 (4 occurrences)
Last logged: 8:56:39

Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for sems3 doing I/O at custom_components/sems3/sensor.py, line 63: _LoginURL, headers=login_headers, data=login_data, timeout=_RequestTimeout)
Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for sems3 doing I/O at custom_components/sems3/sensor.py, line 98: _PowerStationURL, headers=headers, data=data, timeout=_RequestTimeout)

TimSoethout, 2020-05-29:
I have to fix those IO warnings still, but they should not keep it from working.
This is indeed the output I expected.

It should have the two new devices with the SN's as names now. Weird that you don't get them. Since your log indicate it should be working as expected (for now).

spaya1, 2020-05-29:
A new reboot, and they show up :)

Are these names end state? Then I can start building my sensors around them :)

Thanks!! 👍

TimSoethout, 2020-05-29:
Nice. Probably I will make the names something like
sems_{plantName}_{inverterName}.

Does that make sense?

spaya1, 2020-05-29:
Yes, it does! 

Also because I cannot show attributes for the new sensors, and I dont know if its related to the name.

In template:
{{states.sensor.92000ssn18bwXXXX.attributes.pac}} should show 1400 

but shows 

Error rendering template: TemplateSyntaxError: expected token 'end of print statement', got 'ssn18bwXXXX'

TimSoethout, 2020-05-29:
I don't know why that doesn't work, this seems to work: `{{states.sensor['XXXSNXXX'].attributes.pac}}`

TimSoethout, 2020-05-29:
Note that the state of the sensor now represents the `pac` instead of the on/off status:

`{{states.sensor['XXXSNXXX'].state}} == {{states.sensor['XXXSNXXX'].attributes.pac}}`

spaya1, 2020-05-29:
Ah that is great. I see that there is someone else facing something similar:

https://community.home-assistant.io/t/issue-with-template-in-binary-sensor/8687/8

So I will wait until the new sensors names are merged and will use those :)

spaya1, 2020-05-30:
Hi Tim,

I managed to do what I wanted 👍 

I had to do a bit of setup. 

Eg:

sensors:

pv1_outputpower:
        value_template: "{{ (states.sensor['92000ssn18xxxx'].state | float) | round(2) /1000 }}"
        unit_of_measurement: 'kW'
        friendly_name: "PV1 Power Generation Now"
      pv2_outputpower:
       value_template: "{{ (states.sensor['52000ssx202xxxx'].state | float) | round(2) /1000 }}"
       unit_of_measurement: 'kW'
       friendly_name: "PV2 Power Generation Now"
      pv1_outputpowertoday:
        value_template: "{{ (states.sensor['92000ssn18xxxx'].attributes.eday | float) | round(2)}}"
        unit_of_measurement: 'kWh'
        friendly_name: "PV1 Power Generation today"
      pv2_outputpowertoday:
        value_template: "{{ (states.sensor['52000ssx202xxxx'].attributes.eday | float ) | round(2)}}"
        unit_of_measurement: 'kWh'
        friendly_name: "Power Generation today"  
        
Automation (the sensors were not updating for some reason)

  alias: update pvsystem
  description: ''
  trigger:
  - minutes: /1
    platform: time_pattern
  condition: []
  action:
  - data: {}
    entity_id: sensor.pv1_outputpowertoday
    service: homeassistant.update_entity
  - data: {}
    entity_id: sensor.pv2_outputpowertoday
    service: homeassistant.update_entity
  - data: {}
    entity_id: sensor.pv2_outputpower
    service: homeassistant.update_entity
  - data: {}
    entity_id: sensor.pv1_outputpower
    service: homeassistant.update_entity

Result: (Grafana)

[image]

wishie, 2020-09-24:
@spaya1 do you have any batteries? I have found SEMSPortal to make a complete mess of the data when using 2 inverters in 1 plant.. I ended up splitting mine into 2 separate plants.

wishie, 2020-09-24:
@TimSoethout do I need to manually change any files to include 'sems2' in the config etc still, or can I just provide a list of stationids in the sems config?

spaya1, 2021-04-12:
Hi @TimSoethout hope you are fine. I am hoping you can help me. As of yesterday and with the new update of home assistant, custom components where Hassio detects I/O inside the event loop errors mean that the component gets disabled. Updating to the latest version of your component means I lose the access to the second inverter (I get the SEMS Portal sensor pointing to the first inverter of the plant). I was wondering if you could help me regain the functionality you once created for me.

thanks!

spaya1, 2021-04-12:
> @spaya1 do you have any batteries? I have found SEMSPortal to make a complete mess of the data when using 2 inverters in 1 plant.. I ended up splitting mine into 2 separate plants.

Hi, sorry I missed your message, I dont have any batteries, and for some reason for me it works (worked) well. I had the same data in Sems portal as in Hassio, as well as being able to breakdown the data per inverter in the plant.

TimSoethout, 2021-04-12:
Hi, I was writing an async version of the plugin a while back. This might also fix the I/O in event loop issue you're describing. 
If I have time, I will try look at it tonight.

spaya1, 2021-04-12:
> Hi, I was writing an async version of the plugin a while back. This might also fix the I/O in event loop issue you're describing.
> If I have time, I will try look at it tonight.

That would be great, let me know if I can help!

TimSoethout, 2021-04-13:
So, I was trying to work on this tonight, but it already took my whole evening just setting up a hass development environment (using VSCode and docker). Also I'm not sure how I can easily embed this repo in this environment since the docker setup does not seem to support symlinks.
This might take a bit longer than expected...

spaya1, 2021-04-14:
Sorry to hear, I do have my hassio on a docker environment in the rbpi4 let me know if you need me to test something.

Thanks again.

TimSoethout, 2021-04-14:
Yes thanks.
I was trying to set up a proper development setup (separate from my
home/production system) using
https://developers.home-assistant.io/docs/development_environment. Turns
out a lot of things can be wrong and need to be setup before you can
actually start coding. :P

TimSoethout, 2021-04-14:
Ok, I just finished a working version of this on  #10-multiple-inverters+async.
Feel free to test.
If I find some more time I will release a beta version + update documentation.

spaya1, 2021-04-14:
Hi Tim,

thanks for the work, I loaded the branch, and I get the following error:

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:128
Integration: Sensor (documentation, issues)
First occurred: 5:46:48 PM (1 occurrences)
Last logged: 5:46:48 PM

The semstest platform for the sensor integration does not support platform setup. Please remove it from your config.

TimSoethout, 2021-04-14:
It now uses gui config via integrations. I think you need to remove
existing config, and maybe keep only `sems:` to make sure HA picks up the
custom component.

spaya1, 2021-04-14:
Hi Tim,

What I get:

Setup failed for sems: No setup or config entry setup function defined.
6:17:58 PM – (ERROR) setup.py

Logger: homeassistant.setup
Source: setup.py:163
First occurred: 6:17:58 PM (1 occurrences)
Last logged: 6:17:58 PM

Setup failed for sems: No setup or config entry setup function defined.

[image]

[image]

TimSoethout, 2021-04-14:
Your setup seems to be fine. The integration should show up.
[image]

I'm currently trying to also setup again with the code from scratch to see if I missed something.

spaya1, 2021-04-14:
Hi Tim,

I managed to get it to load as a platform. However I get a sensor called SEMS portal where I only see a single inverter.

I used to get 2 sensors, one per inverter.

[image]

spaya1, 2021-04-14:
Hi Tim,

Pretty, sure, is there a way to check in the code?

[image]

TimSoethout, 2021-04-14:
You indeed have the correct code, since you have the config_flow file and translations folder... Hmm.. Not sure what is happening. Is the `sems_portal` sensor maybe an old one still registered in HA?

Can you try removing all of `sems` mentions from your configuration file? It seems it is not needed after all. 
Is it now possible to add the integration via gui?

spaya1, 2021-04-14:
Will try now, but I never had the sems_portal sensor before.

spaya1, 2021-04-14:
Ok so I was able to pull the gun

[image]

But now it picks the second inverter only xD

[image]

TimSoethout, 2021-04-14:
Cool! Progress! You should be able to add the integration twice with the different powerstation id's. At least that was the idea. I can only test with one. :P

spaya1, 2021-04-14:
In my case the plant name is the same for both inverters :). The curious thing is that before (with platform setup) it was showing Inverter 1, and now (with gui setup) it shows Inverter 2.

TimSoethout, 2021-04-14:
Ooooh wait. The `powerstation_id` is the plant id? And not the specific
inverter ID?
If that is so, I need to change the code. :P

spaya1, 2021-04-14:
If you tell me how I can load the api, I can look which are the fields that hold the 2 serial numbers that I am looking for :)

TimSoethout, 2021-04-14:
You can login to https://semsportal.com and in your browsers developer tools on the network tab you can see the different api calls with their JSON responses. Will that help?

spaya1, 2021-04-14:
"is_stored": false,
        "name": "52000",
        "in_pac": 6553.7,
        "out_pac": 22.0,
        "eday": 9.7,
        "emonth": 91.0,
        "etotal": 1785.4,
        "status": 0,
        "turnon_time": "05/26/2020 19:49:46",
        "releation_id": "50e0e175-xxxx",
        "type": "GW2000-XS",
        "capacity": 2.0,
        "d": {
          "pw_id": "5c4e9558-xxxx",
          "capacity": "2kW",
          "model": "GW2000-XS",
          "output_power": "22W",
          "output_current": "0A",
          "grid_voltage": "230.4V",
          "backup_output": "4294967.295V/0W",
          "soc": "200%",
          "soh": "0%",
          "last_refresh_time": "04.14.2021 21:15:09",
          "work_mode": "",
          "dc_input1": "41.4V/0.1A",
          "dc_input2": "0V/0A",
          "battery": "65535V/6553.5A/429483622W",
          "bms_status": "DischargingOfBattery",
          "warning": "Over Temperature/Under Temperature/Cell Voltage Differences/Over Total Voltage/Discharge Over Current/Charge Over Current/Under SOC/Under Total Voltage/Communication Fail/Output Short/SOC Too High/BMS Module Fault/BMS System Fault/BMS Internal Fault/TBD/TBD",
          "charge_current_limit": "1A",
          "discharge_current_limit": "0A",
          "firmware_version": 40410.0,
          "creationDate": "04/15/2021 03:15:09",
          "eDay": 9.7,
          "eTotal": 1785.4,
          "pac": 22.0,
          "hTotal": 3850.0,
          "vpv1": 41.4,
          "vpv2": 0.0,
          "vpv3": 6553.5,
          "vpv4": 6553.5,
          "ipv1": 0.1,
          "ipv2": 0.0,
          "ipv3": 6553.5,
          "ipv4": 6553.5,
          "vac1": 230.4,
          "vac2": 6553.5,
          "vac3": 6553.5,
          "iac1": 0.0,
          "iac2": 6553.5,
          "iac3": 6553.5,
          "fac1": 50.0,
          "fac2": 655.35,
          "fac3": 655.35,
          "istr1": 0.0,
          "istr2": 0.0,
          "istr3": 0.0,
          "istr4": 0.0,
          "istr5": 0.0,
          "istr6": 0.0,
          "istr7": 0.0,
          "istr8": 0.0,
          "istr9": 0.0,
          "istr10": 0.0,
          "istr11": 0.0,
          "istr12": 0.0,
          "istr13": 0.0,
          "istr14": 0.0,
          "istr15": 0.0,
          "istr16": 0.0
        },
        "it_change_flag": false,
        "tempperature": 37.5,
        "check_code": "001201",
        "next": "92000",
        "prev": null,
        "next_device": {
          "sn": "92000",
          "isStored": false
        },
        "prev_device": {
          "sn": null,
          "isStored": false
        },
        "invert_full": {
          "sn": "52000",
          "powerstation_id": "5c4e9558-xxx",
          "name": "52000SSX202W0110",
          "model_type": "GW2000-XS",
          "change_type": 0,
          "change_time": 0,
          "capacity": 2.0,
          "eday": 9.7,
          "iday": 1.94,
          "etotal": 1785.4,
          "itotal": 357.08000000000004,
          "hour_total": 3850.0,
          "status": 0,
          "turnon_time": 1590493786887,
          "pac": 22.0,
          "tempperature": 37.5,
          "vpv1": 41.4,
          "vpv2": 0.0,
          "vpv3": 6553.5,
          "vpv4": 6553.5,
          "ipv1": 0.1,
          "ipv2": 0.0,
          "ipv3": 6553.5,
          "ipv4": 6553.5,
          "vac1": 230.4,
          "vac2": 6553.5,
          "vac3": 6553.5,
          "iac1": 0.0,
          "iac2": 6553.5,
          "iac3": 6553.5,
          "fac1": 50.0,
          "fac2": 655.35,
          "fac3": 655.35,
          "istr1": 0.0,
          "istr2": 0.0,
          "istr3": 0.0,
          "istr4": 0.0,
          "istr5": 0.0,
          "istr6": 0.0,
          "istr7": 0.0,
          "istr8": 0.0,
          "istr9": 0.0,
          "istr10": 0.0,
          "istr11": 0.0,
          "istr12": 0.0,
          "istr13": 0.0,
          "istr14": 0.0,
          "istr15": 0.0,
          "istr16": 0.0,
          "last_time": 1618427709365,
          "vbattery1": 65535.0,
          "ibattery1": 6553.5,
          "pmeter": 22.0,
          "soc": 200.0,
          "soh": -0.100000000000364,
          "bms_discharge_i_max": null,
          "bms_charge_i_max": 1.0,
          "bms_warning": 0,
          "bms_alarm": 65535,
          "battary_work_mode": 2,
          "workmode": 0,
          "vload": 4294967.295,
          "iload": 4294901.76,
          "firmwareversion": 404.0,
          "pbackup": 0.0,
          "seller": 0.0,
          "buy": 0.0,
          "yesterdaybuytotal": null,
          "yesterdaysellertotal": null,
          "yesterdayct2sellertotal": null,
          "yesterdayetotal": null,
          "yesterdayetotalload": null,
          "thismonthetotle": 91.0,
          "lastmonthetotle": 1684.7,
          "ram": 10.0,
          "outputpower": 22.0,
          "fault_messge": 0,
          "isbuettey": false,
          "isbuetteybps": false,
          "isbuetteybpu": false,
          "isESUOREMU": false,
          "backUpPload_S": 0.0,
          "backUpVload_S": 0.0,
          "backUpIload_S": 0.0,
          "backUpPload_T": 0.0,
          "backUpVload_T": 0.0,
          "backUpIload_T": 0.0,
          "eTotalBuy": null,
          "eDayBuy": null,
          "eBatteryCharge": null,
          "eChargeDay": null,
          "eBatteryDischarge": null,
          "eDischargeDay": null,
          "battStrings": 6553.5,
          "meterConnectStatus": null,
          "mtActivepowerR": 0.0,
          "mtActivepowerS": 0.0,
          "mtActivepowerT": 0.0,
          "ezPro_connect_status": null,
          "dataloggersn": "",
          "equipment_name": null,
          "hasmeter": false,
          "meter_type": null,
          "pre_hour_lasttotal": null,
          "pre_hour_time": null,
          "current_hour_pv": 0.0,
          "extend_properties": null,
          "eP_connect_status_happen": null,
          "eP_connect_status_recover": null
        },

TimSoethout, 2021-04-14:
Ok, I just pushed a change that used the inverters serial number as unique id for the HA entity. Maybe this will work already when you re-add the integration.

spaya1, 2021-04-14:
Got two sensors showing now, both for the same inverter, the old one shows unavailable, the new one works and has a new name (the inverter name)

[image]

[image]

spaya1, 2021-04-14:
> Ok, I just pushed a change that used the inverters serial number as unique id for the HA entity. Maybe this will work already when you re-add the integration.

how about using releation_id as unique identifier?

TimSoethout, 2021-04-14:
Yes, I changed some stuff with unique id's and display names. HA might show some old id's/names as well.

I just released a version based on serial number: https://github.com/TimSoethout/goodwe-sems-home-assistant/releases/tag/3.1.0-beta2
This one should also work with HACS if you turn beta versions on. 
Now I'm off to bed. I hope this works. :)

About the `releation_id`, that might also work, as long as it's unique per inverter.

[6 more replies]

closed 2021-04-16 by TimSoethout