TimSoethout/goodwe-sems-home-assistant #12: Unable to fetch data from SEMS. 'NoneType' object is not subscriptable

andyshutak, 2020-01-21:
Hi I've just noticed some errors and the SEMS integration stopping recently. I get the following 2 errors Unable to fetch data from SEMS. 'NoneType' object is not subscriptable and Unable to fetch data from SEMS. Expecting value: line 1 column 1 (char 0). I'm running the latest code, API version. Any help would be greatly appreciated, cheers.

JoLambinon, 2020-01-21:
Hallo all.  Mine is only updates if I restart HA

TimSoethout, 2020-01-22:
For me it still works, so it seems to be specific for your installation. Can you show more of the error trace from the logs?
Or maybe turn on debug logging for this component and share what it logs?

```
logger:
  default: info
  logs:
    custom_components.sems: debug
```
See https://www.home-assistant.io/integrations/logger/.

andyshutak, 2020-01-22:
You're right it worked absolutely fine today, had a load of errors last night but between then and now and a couple of restarts later all is working! Thank you for an excellent plugin! Just need some more sun now lol! :)

closed 2020-01-22 by andyshutak

JoLambinon, 2020-01-23:
Thank you, but mine is also working fine today.