TimSoethout/goodwe-sems-home-assistant #16: Would love to help extend this

wishie, 2020-09-01:
I have 2 x GW5048-EM Hybrid inverters. One with a battery, the other without (for now). I would like to try and assist in getting better/more data about battery states etc. I wrote a small API to power an Android app in my house, and got some insight into how (badly) the data is displayed from the SEMS Portal.

TimSoethout, 2020-09-01:
Please see the experimental issue and branch for multiple inverters: #10

A starting point to extract better data is to look at the JSON the SEMS API supplies, but you might already be doing this.

TimSoethout, 2020-09-01:
And of course pull requests are welcome. 
Also discussions on how to make the component better/ more official-hass-like are nice. :)

wishie, 2020-09-01:
I will take a look into the "multiple inverters" branch.  Are you basing it around 2 inverters in 1 plant, or 2 plants? Ive been down the 2 inverters in 1 plant path, and there are massive issues with how the data is shown/calculated, especially if there are battery/batteries in the mix.

Up to this point, Goodwe have seemed unwilling to help, with their only suggestion being "put the inverters in different plants".

So I wrote my own basic API (to feed my Android app) to help avoid some of their data issues.

What I am mostly interested in, is the 'power flow' data. Showing values for load, pv, battery, grid and the direction (buying/selling/generating etc) of the power.

My ultimate goal (although I currently have no idea how to do this) would be to make a custom Lovelace card that shows data flow.

Here are some screenshots of the Android app I wrote, to show in the middle of the house so the kids can easily understand what's going on. https://imgur.com/a/UfKvGvU

The lines between the solar/load/battery/grid are all animated in the direction of flow at any given time.

TimSoethout, 2020-09-02:
> I will take a look into the "multiple inverters" branch. Are you basing it around 2 inverters in 1 plant, or 2 plants? Ive been down the 2 inverters in 1 plant path, and there are massive issues [...]

We use "_PowerStationURL = 'https://www.semsportal.com/api/v1/PowerStation/GetMonitorDetailByPowerstationId'" for each power station, so I think this is per inverter, independent of plants.

> What I am mostly interested in, is the 'power flow' data. Showing values for load, pv, battery, grid and the direction (buying/selling/generating etc) of the power.
>
> My ultimate goal (although I [...]

Looks really cool. I think you should be able to do something like this using HTML/CSS in lovelace, although I don't have any specific experience with this.

wishie, 2020-09-02:
My system is currently set up as 2 'plants' with 1 inverter in each. This is still not ideal, as the SEMS Portal makes a mess of it, but less than if you have 2 inverters in 1 plant.

Sadly, I am not that good with python. In the 'powerflow' section, the pvStatus/gridStatus/loadStatus/betteryStatus (yes, bettery..) are returned as 1 or -1. In my PHP code I make it return 'generating' or 'selling' or 'idle' etc, based on that, but I am unsure how to do that in python.

TimSoethout, 2020-09-02:
I think you can visualize that better with home assistant templates. You
include some logic to show -1 as "off" and +1 as "generating". See the home
assistant documentation for more info. I think these templates are also
what you use in a lovelace card.

wishie, 2020-09-02:
Sorry, you're right.. I was trying to use the template in configuration.yaml to set the value.. perhaps I was doing it wrong.. will try to read some more documentation

wishie, 2020-09-24:
Ok, I am back on deck and have a few hours to look into multiple inverters. I should also mention, there is now a GoodWe integration that uses UDP directly to the inverter (avoiding SEMSPortal) but not all of the same data is available. I will likely only need the SEMS integration for a few unavailable datapoints via UDP.

TimSoethout, 2021-04-16:
Nice. I would also rather use the UDP plugin, but my inverter model does not seem to be supported yet.

I'll close this issue for now. Feel free to reopen or open a new one. :)

closed 2021-04-16 by TimSoethout