TimSoethout/goodwe-sems-home-assistant #17: Warning "Detected I/O inside the event loop" showing in log using new BETA version

dutchdevil83, 2020-09-01:
Problem: 
```
2020-09-01 16:44:42 WARNING (MainThread) [homeassistant.util.async_] Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for sems doing I/O at custom_components/sems/sensor.py, line 63: login_response = requests.post(
2020-09-01 16:44:42 WARNING (MainThread) [homeassistant.util.async_] Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for sems doing I/O at custom_components/sems/sensor.py, line 98: response = requests.post(
```
Fix:
Add/Wrap those calls as a job to the executor and await hem.

[https://developers.home-assistant.io/docs/asyncio_working_with_async#calling-sync-functions-from-async](https://developers.home-assistant.io/docs/asyncio_working_with_async#calling-sync-functions-from-async)

Please note, that the current state of this integration make Home Assistant execution halt/blocked until these calls are finished. Basically, this integration causes hiccups for the whole system now.

Looking at the error logs i think code line 105 needs editing:
`            response = requests.post(_PowerStationURL, headers=headers, data=data, timeout=_RequestTimeout)`

TimSoethout, 2020-09-01:
Hmm.. I tagged the release as pre-release. Does HACS automatically push for update then? I was hoping it would not.

TimSoethout, 2020-09-01:
If so, I will remove the release again until it is stable enough.

TimSoethout, 2020-09-01:
However, I of course I agree this has to be fixed before proper release. Thanks for reporting and pointing to a fix.

Sverto, 2021-02-09:
You could use 
`import requests_async as requests`
+ add 'await' requests.post(...
+ make functions 'async' def ...

But it requires the requests_async lib to be present.
(pip install requests_async)

TimSoethout, 2021-04-13:
Thanks @Sverto, that seems like a good quick fix. Maybe we should try that first instead of the total rewrite to async.

TimSoethout, 2021-04-16:
Version 3.1.0 now uses async correctly. Thanks for reporting. It should be fixed now. Let me know if it is not.

closed 2021-04-16 by TimSoethout