TimSoethout/goodwe-sems-home-assistant #18: Warnings and errors in Home Assistant log

Bartvdgriendt, 2020-09-03:
I've been using this integration for a couple weeks now and I think it's great. Thanks! 

Today I noticed an error and a warning in the Home Assistant logs. Both occur quite frequently and I can't find a particular pattern.
`2020-09-01 13:18:05 ERROR (SyncWorker_18) [custom_components.sems.sensor] Unable to fetch data from SEMS. HTTPSConnectionPool(host='www.semsportal.com', port=443): Read timed out. (read timeout=30)`
and
`2020-09-01 09:37:45 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.sems_portal is taking over 10 seconds`

I found these in Supervisor>System>System Log> Core tab. I don't experience any problems and the data from Goodwe is still updated, only sometimes delayed. Is this an error on my side or maybe a bug in the integration? I couldn't find much info online.

Extra info:
- Home Assistant installed as a VM on my ProxMox server
- Home Assistant version 0.114.4
- Goodwe SEMS integration for Home Assistant version 1.0.0 (installed with HACS)

If any extra info is needed, please let me know!

TimSoethout, 2020-09-03:
Hi @Bartvdgriendt, I've checked it for myself (still running HA Core 0.111.4), but I don't have the entries, although I remember seeing them sometimes. I don't think they do any harm, except for maybe some missing values in your history.

It seems that the portal is taking too long to respond. This can have multiple reasons: your connection to SEMS is down/slow, SEMS servers are slow/down, maybe you're rate limited by the SEMS API, something else?
Are you using the default scan interval (`scan_interval: 60`)? I have not tested the component with lower values.

You can try pinging the SEMS API from your system and see if latency increases around the same time as the timeouts happen? Maybe just using ping in a terminal, or using smokeping. If you really want to test HTTPS latency maybe use `echoping`?

TimSoethout, 2020-09-03:
Also remember the API/website misbehaving when at the same time using their webportal, but that might be from the time we didn't use the API, but were still scraping their website.

Bartvdgriendt, 2020-09-05:
@TimSoethout Thanks for the quick response and the explanation. I too don't think they do any harm but I always get nervous when I see errors in my logs. ;) I ruled out problems on my side:

- My connection to SEMS is down/slow: I have a 200/200 connection and I run different services on the same server which need internet access 24/7. When internet access is down/slow I'll receive notifications from local and remote devices. I didn't receive any at the time of the errors.
- I changed the scan_interval to 600 because I don't need it to be updated every minute but this doesn't seem to help.

My best guess is that the SEMS API isn't very reliable but then others should have the same errors. I'll try to ping the website for a day and check if I see something useful when I receive the errors. I'm not sure when but I will keep you updated!

dutchdevil83, 2020-09-18:
can confirm that i have those errors in the log also. Probaly API not responding, no big issue if you ask me.

Bartvdgriendt, 2020-09-19:
Great timing @dutchdevil83 :) I just finished pinging the website for 2 days. I did that from the ProxMox server that runs HA and did that continuously. 

I compared the results with the errors in HA and when those errors occur the latency of the ping spikes and sometimes the host can't even be reached. So as dutchdevil83 and @TimSoethout already said/suspected: it's probably the API that isn’t very reliable. I assume you're both Dutch (based on your usernames) so maybe we can compare the errors to be 100% sure? My most recent error occurred ~45 minutes ago:

```
2020-09-19 14:23:05 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.sems_portal is taking over 10 seconds
2020-09-19 14:23:25 ERROR (SyncWorker_31) [custom_components.sems.sensor] Unable to fetch data from SEMS. HTTPSConnectionPool(host='www.semsportal.com', port=443): Read timed out. (read timeout=30)
```

TimSoethout, 2020-09-21:
```
2020-09-19 14:23:10 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.sems_portal is taking over 10 seconds
2020-09-19 14:23:30 ERROR (SyncWorker_20) [custom_components.sems.sensor] Unable to fetch data from SEMS. HTTPSConnectionPool(host='www.semsportal.com', port=443): Read timed out. (read time
out=30)
2020-09-19 14:24:10 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.sems_portal is taking over 10 seconds
2020-09-19 14:24:30 ERROR (SyncWorker_0) [custom_components.sems.sensor] Unable to fetch data from SEMS. HTTPSConnectionPool(host='www.semsportal.com', port=443): Read timed out. (read timeo
ut=30)
```
Yup also here on the same minute. Seems to happen more frquently now.

TimSoethout, 2020-09-21:
I don't think we can fix the time-outs themselves.
What we can do it label them differently. Time-outs can be warnings, and maybe an error or notification if it happens to often or something.
What do you think?

Bartvdgriendt, 2020-09-22:
Thanks, that confirms that the SEMS API isn’t 100% reliable and that it's not a problem on my side. 

I too don't think we can fix the time-outs. In fact: I _know_ we can't do anything about them. ;) I think that your solution (making time-outs warnings and make them errors when they occure too often) is a great solution. Besides that you could add some information about the unreliable API in the Wiki/Readme to inform users that these warnings/errors are ‘normal’ and that they can't do anything about them.

TimSoethout, 2021-04-16:
I added a note on this in the readme for now.

closed 2021-04-16 by TimSoethout