TimSoethout/goodwe-sems-home-assistant #22: Battery and PV levels?

HeinTheTerrible, 2021-01-07:
Is it possible to scrap battery level and current PV generation data?

I tried the following for the battery but it didn't work:
      pv_ibattery:
        value_template: '{{ states.sensor.sems_portal.attributes.ibattery }}'
        unit_of_measurement: 'kWh'
        friendly_name: "Battery level"

Thanks for the SEMS sensor so far! Works great!

TimSoethout, 2021-01-07:
Hey, unfortunately I don't have a battery, so I'm not sure which fields to use for this?

Maybe you can try to look at the json?
Log in to: https://www.semsportal.com/
Open de develop tools en check the json of `https://www.semsportal.com/GopsApi/Post?s=v1/PowerStation/GetMonitorDetailByPowerstationId` in the network tab.

HeinTheTerrible, 2021-01-08:
Hi, seems like Goodwe has blocked access to the API sites. Most of the other links I found didn't work either.

I did try a couple of different attributes and got the battery SOC to show!

      pv_soc:
        value_template: '{{ states.sensor.sems_portal.attributes.soc }}'
        unit_of_measurement: '%'
        friendly_name: "Battery power"

Currently, the "power output" sensor shows the combined output of the PV panels and battery and NOT the PV output only.

      pv_outputpower:
        value_template: '{{ states.sensor.sems_portal.attributes.outputpower }}'
        unit_of_measurement: 'W'
        friendly_name: "PV Power output"

As you mentioned before you don't have a battery to test and I'm a complete noob with this stuff. Thought I just share my findings.

Cheers

TimSoethout, 2021-02-08:
Thanks. I added them as examples to the readme. :)

closed 2021-02-08 by TimSoethout

rastislavpapp, 2021-04-05:
Hi, the battery power value is located in the `/GetMonitorDetailByPowerstationId` response  at the `jsonResponseFinal["data"]["inverter"][0]` level as `battery_power` (in Watts):

```
{
  ...
  "data": {
    ...
    "inverter": [
      {
        "invert_full": {
          ...
        },
        ...
        "battery_power": 20.17,
        ...
      }
    ]
  }
}
```

so adding:

```
self._attributes["Pbattery"] = jsonResponseFinal["data"]["inverter"][0]["battery_power"]
```
at the end of `sensor.py` would solve this.