TimSoethout/goodwe-sems-home-assistant #24: When sun goes down PV sensor does not go to 0 Watt

lukaskrepel, 2021-02-12:
Hi, thanks for the sensor!

When the sun goes down the PV power sensor does not go to 0 Watt, but instead hangs on the last measured value.

spockz, 2021-02-12:
I have this issue as well on the SEMS portal. I assume it is a bug on the api side where it keeps using the latest reported value in case of a missed shutdown message from the PV. (The last part is speculation.)

TimSoethout, 2021-02-13:
We could divert from the API output by overwrite the PV power sensor to `0`
when the inverter is off (found in another attribute). However, I don't
know if this is correct in all cases, e.g. if you have batteries?

lukaskrepel, 2021-02-13:
Sounds like a good solution for my case! I don't have a battery...
I would assume the output would be zero for PV even with batteries, but I am not familiar with the SEMS portal API.

lukaskrepel, 2021-02-13:
By the way: My current solution is to manually restart HA when the sun sets. The sensor goes to zero Watts. So I think the API simply doesn't respond when there's no power?

TimSoethout, 2021-04-16:
In version 3.1.0 I added a check to report 0W when `status` is not `1`. :)

closed 2021-04-16 by TimSoethout