TimSoethout/goodwe-sems-home-assistant #25: HA 2021.3.0 requests version manifest

pvmil, 2021-03-04:
As described in releasenotes of HA 2021.3.0, a version number is required in the manifest file: 
https://www.home-assistant.io/blog/2021/03/03/release-20213/#read-more

Output HA log:
Logger: homeassistant.loader
Source: loader.py:802
First occurred: 17:32:12 (2 occurrences)
Last logged: 17:32:13

No 'version' key in the manifest file for custom integration 'sems'. This will not be allowed in a future version of Home Assistant. Please report this to the maintainer of 'sems'

Ascariota, 2021-03-04:
I have the same :(

TimSoethout, 2021-03-07:
Thanks for mentioning. I added a version to the manifest and hope it works. (Did not test yet.)

closed 2021-03-07 by TimSoethout

pvmil, 2021-03-07:
Tested, this works, thnx!