TimSoethout/goodwe-sems-home-assistant #27: Integration setup hangs

nickh66, 2021-04-19:
Hi,

I have not been a long time fan of your custom integration.  It has worked very well for so far but with this new upgrade i am now not able to use the new 3.1.1 version.  The HACS integration upgrade went well (reinstallations also installing normally) & the HA integration removal/reinstall also seems to add normally.  

However, after the HA integration is added (also if i restart HA) i get these series of errors that seems to hang up some other parts of my HA instance (both prd & dev instances behave the same) & other integrations like automations don't work normally.  Also, i never see any sensors created.

These are the errors i see in my log.

Also & a perhaps an indication of something (not sure what though) i found that my sems portal web account seems to be getting 'expired' now.  Until i remove the integrations & restart to stop continued setup attempts i found i could not even log on to the website with out an 'login expired' message.  After the integration was removed i could once again use the sems portal website normally.

Hope this information helps you to troubleshoot thins & i am more than happy to work with you on this issue if that is of any help.

Regards
Nick Harrison
-----------------------------------------------------------------------------------------------------------------------
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:237
Integration: Sensor (documentation, issues)
First occurred: 9:34:06 (1 occurrences)
Last logged: 9:34:06

Setup of platform sems is taking longer than 60 seconds. Startup will proceed without waiting any longer.
----------------------------------------------------------------------------------------------------------------------
Logger: homeassistant.bootstrap
Source: bootstrap.py:554
First occurred: 9:38:12 (1 occurrences)
Last logged: 9:38:12

Setup timed out for bootstrap - moving forward
---------------------------------------------------------------------------------------------------------------------
Logger: custom_components.sems.sems_api
Source: custom_components/sems/sems_api.py:71
Integration: GoodWe SEMS API (documentation, issues)
First occurred: 9:40:31 (2 occurrences)
Last logged: 9:41:03

Unable to fetch login token from SEMS API. HTTPSConnectionPool(host='eu.semsportal.com', port=443): Read timed out. (read timeout=30)

lowgoz, 2021-04-20:
Have same problem here :(

TimSoethout, 2021-04-20:
Thanks for reporting this.
I made some other changes in the new version, one of them could be the culprit:
- `eu.semsportal.com` instead of `www.semsportal.com`, although for me they always resolve to the same ip, maybe from Australia they don't?
- v2 of the api instead of v1 (+ some change of HTTP headers)
- Trying to reuse old token and repeating if token is invalid. Although the number of retries is is unlimited in the code right now, so that also might result in unlimited failed retries in your case?

> Also & a perhaps an indication of something (not sure what though) i found that my sems portal web account seems to be getting 'expired' now. Until i remove the integrations & restart to stop [...]

I also notices this. When the plugin logs in, it logs you out of the website (and maybe also app). This could be because of the v2 API? Or maybe something else.

Can you try turning on debug logging in your `application.yaml` and showing the resulting logs?
```yaml
logger:
  default: info
  logs:
    custom_components.sems: debug # only add this line if logger is already present
```

kaizersoje, 2021-04-20:
Same problem here.

nickh66, 2021-04-21:
Hey Tim,

Thanks for following up on this.  Much appreciated.

So i re-installed the integration on my dev instance let it soak for a while with the debug logging enabled.  Wow, lots of detail now in the full log.  Very frequent log entries too.  This is just a sample of the log information but this just repeats every few seconds it seems.

2021-04-21 12:10:55 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Getting API token
2021-04-21 12:11:00 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Login Response: <Response [200]>
2021-04-21 12:11:00 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - API Token received: {"uid": "08a90239-d259-446c-8be3-3bce23f941ff", "timestamp": 1618971060118, "token": "f5df8531e3cc108ef8894e1c7dbaa825", "client": "web", "version": "", "language": "en"}
2021-04-21 12:11:00 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Querying SEMS API for power station id: 3d67ede6-0295-4bdb-a80a-e7aa43395080
2021-04-21 12:11:02 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Query not successful (The authorization has expired, please login again.), retrying with new token
2021-04-21 12:11:02 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-04-21 12:11:02 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] API token not set ({"uid": "08a90239-d259-446c-8be3-3bce23f941ff", "timestamp": 1618971060118, "token": "f5df8531e3cc108ef8894e1c7dbaa825", "client": "web", "version": "", "language": "en"}) or new token requested (True), fetching

2021-04-21 12:11:02 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Getting API token
2021-04-21 12:11:07 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Login Response: <Response [200]>
2021-04-21 12:11:07 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - API Token received: {"uid": "08a90239-d259-446c-8be3-3bce23f941ff", "timestamp": 1618971067446, "token": "e6ec005ff1255ea725d76567e95280ee", "client": "web", "version": "", "language": "en"}
2021-04-21 12:11:07 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Querying SEMS API for power station id: 3d67ede6-0295-4bdb-a80a-e7aa43395080
2021-04-21 12:11:09 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Query not successful (The authorization has expired, please login again.), retrying with new token
2021-04-21 12:11:09 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-04-21 12:11:09 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] API token not set ({"uid": "08a90239-d259-446c-8be3-3bce23f941ff", "timestamp": 1618971067446, "token": "e6ec005ff1255ea725d76567e95280ee", "client": "web", "version": "", "language": "en"}) or new token requested (True), fetching

2021-04-21 12:11:09 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Getting API token
2021-04-21 12:11:14 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Login Response: <Response [200]>
2021-04-21 12:11:14 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - API Token received: {"uid": "08a90239-d259-446c-8be3-3bce23f941ff", "timestamp": 1618971074540, "token": "68596de3001dc3453f2bcae06b68f077", "client": "web", "version": "", "language": "en"}
2021-04-21 12:11:14 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Querying SEMS API for power station id: 3d67ede6-0295-4bdb-a80a-e7aa43395080
2021-04-21 12:11:17 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Query not successful (The authorization has expired, please login again.), retrying with new token
2021-04-21 12:11:17 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-04-21 12:11:17 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] API token not set ({"uid": "08a90239-d259-446c-8be3-3bce23f941ff", "timestamp": 1618971074540, "token": "68596de3001dc3453f2bcae06b68f077", "client": "web", "version": "", "language": "en"}) or new token requested (True), fetching

Hope this helps.  

Regards
Nick

TimSoethout, 2021-04-21:
Thanks. It indeed seems to try and login every 5 seconds. I guess the API doesn't like that and also logs you out on the webpage.
A quick fix is to limit the number of retries for now.

TimSoethout, 2021-04-21:
I added a maximum number of retries for now. Can you try and install the `master` version from HACS? It should pull in this change.

nickh66, 2021-04-21:
> I added a maximum number of retries for now. Can you try and install the `master` version from HACS? It should pull in this change.

Hi Tim, Yes i will do that today.

nickh66, 2021-04-21:
Hi Tim,

I setup the 'master' version & i can see now that the setup only tries a few times before it aborts.  Still not making connection through the API though.  Also, there seems to be another error right after but i can't be sure it's related or not.

2021-04-21 19:02:11 INFO (SyncWorker_12) [custom_components.sems.sems_api] SEMS - Maximum token fetch tries reached, aborting for now
2021-04-21 19:02:11 DEBUG (MainThread) [custom_components.sems.sensor] Finished fetching SEMS API data in 15.322 seconds
2021-04-21 19:03:14 ERROR (MainThread) [aiohttp.server] Unhandled exception
Traceback (most recent call last):
File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1152, in _sendfile_fallback
read = await self.run_in_executor(None, file.readinto, view)
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 485, in start
resp, reset = await task
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 440, in _handle_request
reset = await self.finish_response(request, resp, start_time)
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 591, in finish_response
await prepare_meth(request)
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_fileresponse.py", line 241, in prepare
return await self._sendfile(request, fobj, offset, count)
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_fileresponse.py", line 96, in _sendfile
await loop.sendfile(transport, fobj, offset, count)
File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1131, in sendfile
return await self._sendfile_fallback(transport, file,
File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1161, in _sendfile_fallback
await proto.restore()
File "/usr/local/lib/python3.8/asyncio/base_events.py", line 263, in restore
self._transport.resume_reading()
File "/usr/local/lib/python3.8/asyncio/sslproto.py", line 344, in resume_reading
self._ssl_protocol._transport.resume_reading()
AttributeError: 'NoneType' object has no attribute 'resume_reading'

lowgoz, 2021-04-21:
> Thanks for reporting this.
> I made some other changes in the new version, one of them could be the culprit:
> 
> * `eu.semsportal.com` instead of `www.semsportal.com`, although for me they always resolve to the same ip, maybe from Australia they don't?

looks like they do not ! 
[image]

TimSoethout, 2021-04-21:
> > Thanks for reporting this.
> > I made some other changes in the new version, one of them could be the culprit:
> > 
> > * `eu.semsportal.com` instead of `www.semsportal.com`, although for me they always resolve to the same ip, maybe from Australia they don't?
> 
> looks like they do not !
> [image]

Interesting! I changed the it back to `www` on master. Also using `ios` instead of `web` in HTTP headers now. It seems to have less issues with expired logins.

nickh66, 2021-04-22:
Well done Tim.  Today after reinstalling 'master' again to pick up this latest change i found that the setup completes almost instantly now with no errors & my inverter is found & showing me data again.

I will keep the debug logging turned on for now & monitor over the next day or so.  Should i stay on 'master' or will you implement these changes to 3.1.x?

Also, is there a way in this new UI based integration to change the polling interval out to the cloud.  Right now from the logs it looks like it's polling every 1 minute.  For my needs i would be happy with every 5+.  It was a nice feature in the older version to have the ability to customise that timing.

Cheers
Nick

nickh66, 2021-04-22:
Hi Tim,

Got home from the office today & just checked the logs for errors.  Found this error.  Don't know if it's related to sems or not but thought i would share just in case.

Logger: aiohttp.server
Source: /usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py:393
First occurred: 18:13:15 (1 occurrences)
Last logged: 18:13:15

Unhandled exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1152, in _sendfile_fallback
    read = await self.run_in_executor(None, file.readinto, view)
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 485, in start
    resp, reset = await task
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 440, in _handle_request
    reset = await self.finish_response(request, resp, start_time)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 591, in finish_response
    await prepare_meth(request)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_fileresponse.py", line 241, in prepare
    return await self._sendfile(request, fobj, offset, count)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_fileresponse.py", line 96, in _sendfile
    await loop.sendfile(transport, fobj, offset, count)
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1131, in sendfile
    return await self._sendfile_fallback(transport, file,
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1161, in _sendfile_fallback
    await proto.restore()
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 263, in restore
    self._transport.resume_reading()
  File "/usr/local/lib/python3.8/asyncio/sslproto.py", line 344, in resume_reading
    self._ssl_protocol._transport.resume_reading()
AttributeError: 'NoneType' object has no attribute 'resume_reading'

TimSoethout, 2021-04-22:
Could be. Although there seems to be no direct link. It can be that the sems code returns `None` somewhere in an error scenario.

Is there any surrounding SEMS debug logging present?

P.S. Will look later at configuring the the update interval later. Issue #29 as reminder.

nickh66, 2021-04-22:
Hi Tim,

This is a snippet of log from just now as i logon this morning to take a look.  

2021-04-23 08:29:50 DEBUG (SyncWorker_11) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-04-23 08:29:50 DEBUG (SyncWorker_11) [custom_components.sems.sems_api] Querying SEMS API for power station id: 3d67ede6-0295-4bdb-a80a-e7aa43395080
2021-04-23 08:29:50 DEBUG (MainThread) [custom_components.sems.sensor] Found inverter attribute GW5000L-DT 55000DTU193W0031
2021-04-23 08:29:50 DEBUG (MainThread) [custom_components.sems.sensor] Finished fetching SEMS API data in 0.263 seconds
2021-04-23 08:29:57 ERROR (MainThread) [aiohttp.server] Unhandled exception
Traceback (most recent call last):
File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1152, in _sendfile_fallback
read = await self.run_in_executor(None, file.readinto, view)
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 485, in start
resp, reset = await task
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 440, in _handle_request
reset = await self.finish_response(request, resp, start_time)
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 591, in finish_response
await prepare_meth(request)
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_fileresponse.py", line 241, in prepare
return await self._sendfile(request, fobj, offset, count)
File "/usr/local/lib/python3.8/site-packages/aiohttp/web_fileresponse.py", line 96, in _sendfile
await loop.sendfile(transport, fobj, offset, count)
File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1131, in sendfile
return await self._sendfile_fallback(transport, file,
File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1161, in _sendfile_fallback
await proto.restore()
File "/usr/local/lib/python3.8/asyncio/base_events.py", line 263, in restore
self._transport.resume_reading()
File "/usr/local/lib/python3.8/asyncio/sslproto.py", line 344, in resume_reading
self._ssl_protocol._transport.resume_reading()
AttributeError: 'NoneType' object has no attribute 'resume_reading'
2021-04-23 08:30:50 DEBUG (SyncWorker_7) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-04-23 08:30:50 DEBUG (SyncWorker_7) [custom_components.sems.sems_api] Querying SEMS API for power station id: 3d67ede6-0295-4bdb-a80a-e7aa43395080
2021-04-23 08:30:50 DEBUG (MainThread) [custom_components.sems.sensor] Found inverter attribute GW5000L-DT 55000DTU193W0031
2021-04-23 08:30:50 DEBUG (MainThread) [custom_components.sems.sensor] Finished fetching SEMS API data in 0.903 seconds

Cheers
Nick

nickh66, 2021-04-26:
Thanks for the updates Tim.  All working great again.
Cheers
Nick

closed 2021-04-26 by nickh66