TimSoethout/goodwe-sems-home-assistant #28: Sensor name is 'sensor.inverter_none'

bitcrumb, 2021-04-22:
Hi,

after reinstalling the integration since the 3.1.0 update (I think), the inverter is now exposed through a sensor called `sensor.invertor_none`. Previously this was `sensor.sems_portal`.

This looks fishy to me. I expected to see the name of my inverter as defined on the semsportal.com website (`Management > Plant > Device Management > Modify`. I explicitly gave my inverter a name there after observing that the sensor in home assistant reported "none" as the name.

However, I don't see the sensor being renamed...

bitcrumb, 2021-04-22:
Ah, I just read the other issue (https://github.com/TimSoethout/goodwe-sems-home-assistant/issues/27) where you gave instructions on how to debug by adding extra logs.

It seems the API keeps returning `None` as the name for the inverter in the `invert_full` playload:

```
"invert_full":{
   "sn":"13000DSN164W0184",
   "powerstation_id":"da550800-01de-4062-9337-03f193b5b24e",
   "name":"None", <------------------------------------------------------------ 👀
   "model_type":"GW3000D-NS"
   ...
}
```

TimSoethout, 2021-04-22:
Thanks for reporting. 
It should indeed show the field you're referencing for the name. Home assistant should pick a unique name if there is a clash, just as `Light 2` for example. However I did not test this, since I only have a single inverter. (Although mocks and unit tests would allow us to test this of course...)
The unique id, is the serial number, which is what HA uses internally for uniqueness.

As a work around you can always change the device id and name in the HA gui yourself. :) Since it is internally using SN, it should update just fine. :)

bitcrumb, 2021-04-22:
That's a good suggestion indeed, I'll change it via HA.

Guess I am not surprised about the API behaving like this... the Goodwe SEMS API & portal seem to be of really poor quality unfortunately :(

Thanks again :)

TimSoethout, 2021-04-22:
Great! I will close this issue for now, since I don't know if we can fix this on our side. :)

closed 2021-04-22 by TimSoethout