TimSoethout/goodwe-sems-home-assistant #29: Make update interval configurable

TimSoethout, 2021-04-22:


jwolbink, 2021-04-24:
Hello Tim, After upgrade I only see the Inverter but not the other sensors. Please let me know how to solve it. Greetings , Johan.

TimSoethout, 2021-04-24:
Hallo Johan, 
I guess this is not really related to this specific issue, but which other sensors do you mean?
The templates as mentioned in the readme should also be updated in your `application.yaml` since the sensor name/id is different. Does this help?

jwolbink, 2021-04-24:
In the integration i see only one entity (sensor.inverter........). I'm missing the outputpower, tempperature, eday, etotal etc. I'm new so maybe it's not much what i'm doing wrong.

TimSoethout, 2021-04-24:
Hi, if you open the entity, the other fields shoeld be visible as attributes:
[image]

jwolbink, 2021-04-24:
Yes, Thats correct. How can I see some of this values in HA ?

TimSoethout, 2021-04-25:
You need to replace $NAME with the name of your sensor. The `sensor.sems_portal` you mention no longer exists, since we support multiple inverters now.  Look in `http://.../config/entities` for you entity and use the new sensor id. :) Hope it works.

TimSoethout, 2021-04-25:
@nickh66, update interval should now be configurable in config flow.

TimSoethout, 2021-04-25:
I'll release a new release if it works for you.

jwolbink, 2021-04-25:
> You need to replace `$NAME` with the name of your sensor. The `sensor.sems_portal` you mention no longer exists, since we support multiple inverters now. Look in `http://.../config/entities` for you entity and use the new sensor id. :) Hope it works.

> I have replace   $NAME . What do tou mean with: Look in `http://.../config/entities` for you entity and use the new sensor id ?

> btw. please open a new issue, if this doesn't work.
> 
> This issue actually for a different feature request. ;)

TimSoethout, 2021-04-29:
Released as version 3.2.0

closed 2021-04-29 by TimSoethout