TimSoethout/goodwe-sems-home-assistant #30: Do not show attributes

Haplo1577, 2021-04-26:
Dear sir,
After I have updated your integrations, which worked perfectly before, I cannot use the attributes for the inverter in a card. If I open the entity, I can see the attributes like you show in the readme text.
I have changed the name in the template in configuration.yaml but nothing happens.

sensor:
    - platform: template
      sensors:
        pv_outputpower:
          value_template: '{{ states.sensor.inverter_atahualpa.attributes.outputpower }}'
          unit_of_measurement: 'W'
          friendly_name: "Producción de las placas"
        pv_temperature:
          value_template: '{{ states.sensor.inverter_atahualpa.attributes.tempperature }}'
          unit_of_measurement: 'C'
          friendly_name: "Temperatura de las placas"

Do you have any idea about this problem?
Thank you very much in advanced.

TimSoethout, 2021-04-26:
Hi, That should indeed work. Can you see if the templates work, for example via: `http://hassio.lan:8123/developer-tools/template` (replace domain with your HA location)
If I try this (with my inverter name)
`{{ states.sensor.inverter_atahualpa.attributes.outputpower }}`
I see `123`

[image]

You can also try `states.sensor.inverter_atahualpa.attributes` to see all attributes.

Haplo1577, 2021-04-27:
Finally, I got it.

I put this in my template and works.

[image: image.png]
Thank you very much.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

TimSoethout, 2021-04-27:
Great!

closed 2021-04-27 by TimSoethout