TimSoethout/goodwe-sems-home-assistant #31: homeassistant.exceptions.TemplateError: UndefinedError: 'None' has no attribute 'attributes'

TheGabeMan, 2021-04-26:
Hi
Today I update my HA to 2021.4.6 and after reboot I checked the log and saw the following errors:
homeassistant.exceptions.TemplateError: UndefinedError: 'None' has no attribute 'attributes'

and

Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.tempperature }}'
Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.eday }}'
Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.etotal }}'
Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.iday }}'
Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.itotal }}'

I must admit, I'm not sure if these errors have occurred with earlier updates as well, could be these were in it for some time.

Expanded logs:

Logger: homeassistant.helpers.template
Source: helpers/template.py:1346
First occurred: 8:24:39 PM (84 occurrences)
Last logged: 8:36:40 PM

Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.tempperature }}'
Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.eday }}'
Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.etotal }}'
Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.iday }}'
Template variable error: 'None' has no attribute 'attributes' when rendering '{{ states.sensor.sems_portal.attributes.itotal }}'

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:133
Integration: Sensor (documentation, issues)
First occurred: 8:24:16 PM (2 occurrences)
Last logged: 8:28:53 PM

The sems platform for the sensor integration does not support platform setup. Please remove it from your config.
The thinq_v2 platform for the sensor integration does not support platform setup. Please remove it from your config.

Logger: homeassistant.components.template.template_entity
Source: components/template/template_entity.py:72
Integration: Template (documentation, issues)
First occurred: 8:24:40 PM (6 occurrences)
Last logged: 8:24:40 PM

TemplateError('UndefinedError: 'None' has no attribute 'attributes'') while processing template 'Template("{{ states.sensor.sems_portal.attributes.tempperature }}")' for attribute '_state' in entity 'sensor.pv_temperature'
TemplateError('UndefinedError: 'None' has no attribute 'attributes'') while processing template 'Template("{{ states.sensor.sems_portal.attributes.eday }}")' for attribute '_state' in entity 'sensor.pv_eday'
TemplateError('UndefinedError: 'None' has no attribute 'attributes'') while processing template 'Template("{{ states.sensor.sems_portal.attributes.etotal }}")' for attribute '_state' in entity 'sensor.pv_etotal'
TemplateError('UndefinedError: 'None' has no attribute 'attributes'') while processing template 'Template("{{ states.sensor.sems_portal.attributes.iday }}")' for attribute '_state' in entity 'sensor.pv_iday'
TemplateError('UndefinedError: 'None' has no attribute 'attributes'') while processing template 'Template("{{ states.sensor.sems_portal.attributes.itotal }}")' for attribute '_state' in entity 'sensor.pv_itotal'

TheGabeMan, 2021-04-26:
My configuration.yaml contains this:

`  # Optional/example
  # A template to ease access to the data as "sensor.pv_outputpower" etc.
  - platform: template
    sensors:
      pv_outputpower:
        value_template: "{{ states.sensor.sems_portal.attributes.outputpower }}"
        unit_of_measurement: "W"
        friendly_name: "PV Power output"
      pv_temperature:
        value_template: "{{ states.sensor.sems_portal.attributes.tempperature }}"
        unit_of_measurement: "C"
        friendly_name: "PV Temperature"
      pv_eday:
        value_template: "{{ states.sensor.sems_portal.attributes.eday }}"
        unit_of_measurement: "kWh"
        friendly_name: "PV energy day"
      pv_etotal:
        value_template: "{{ states.sensor.sems_portal.attributes.etotal }}"
        unit_of_measurement: "kWh"
        friendly_name: "PV energy total"
      pv_iday:
        value_template: "{{ states.sensor.sems_portal.attributes.iday }}"
        unit_of_measurement: "€"
        friendly_name: "PV income day"
      pv_itotal:
        value_template: "{{ states.sensor.sems_portal.attributes.itotal }}"
        unit_of_measurement: "€"
        friendly_name: "PV income total"`

TimSoethout, 2021-04-26:
Hi, the latest plugin version has different sensor ids to support multiple inverters. Please see: https://github.com/TimSoethout/goodwe-sems-home-assistant#extra-optional-templates-to-easy-access-data-as-sensors

TheGabeMan, 2021-04-26:
Ah, thank you. I never finished the step of adding it in the custom integrations, even though it was working without issues. Now with changing the template as you suggested, it is fixed. Thx

closed 2021-04-26 by TheGabeMan