TimSoethout/goodwe-sems-home-assistant #33: Update Scan Interval

luke2700k, 2021-06-01:
Probably just something I have overlooked, but how can I update the scan interval (without readding the entity)? Its currently set to the default of 60 seconds but I would like to change it to 30 seconds. I have tried updating DEFAULT_SCAN_INTERVAL and CONF_SCAN_INTERVAL with no luck. Thanks in advance for any advice.

TimSoethout, 2021-06-01:
I think you need to re-add the entity/integration in order to set the `scan_interval`. It should be visible in the config flow.
I'm not sure how to do this otherwise, since you can't just change some config yaml anymore.

luke2700k, 2021-06-01:
Hi Tim, thanks for the quick reply (and the awesome GoodWe integration!!) No worries I will just re-add the entity. :)

luke2700k, 2021-06-01:
Issue closed.

closed 2021-06-01 by luke2700k