TimSoethout/goodwe-sems-home-assistant #34: Won't show up in integrations

Gotanius, 2021-06-11:
I've installed through HACS.
Tried 3.2.0 and master. Installation seems to work. Files are available in custom components. HomeAssistant recognizes the fact that custom componen has been installed. 
However, I can't add it through integrations. When I search for it, it shows no results.

wsiczek, 2021-06-17:
I came across similar issue. Did you fix it already?

TimSoethout, 2021-06-18:
master and 3.2.0 should be the same version at the moment.
`configuration.yml` based config does indeed not work anymore.
I'm still on `core-2021.5.1` and it seems to work. Which version of HA do you use? I can try to upgrade to it and see what happens.

Normally you would see:
```
2021-06-15 19:00:12 INFO (MainThread) [homeassistant.setup] Setting up sems
2021-06-15 19:00:12 INFO (MainThread) [homeassistant.setup] Setup of domain sems took 0.0 seconds
```
in the logs, but maybe that is only when a device is already present.

```
2021-06-15 19:00:06 WARNING (MainThread) [homeassistant.loader] You are using a custom integration sems which has not been tested by Home Assistant. This component might cause stability prob
lems, be sure to disable it if you experience issues with Home Assistant
``` 
should always be visible in the logs.

wsiczek, 2021-06-18:
Updating HassOS to 2021.6.5 help to solve the issue in my case :)

Gotanius, 2021-06-18:
Indeed updating to 2021.6.5 fixed it for me as well :)

closed 2021-06-18 by Gotanius

HeinBr, 2021-08-05:
Hello,

I have the same problem. It does not show up in the integrations. I have the latest update core-2021.8.1