TimSoethout/goodwe-sems-home-assistant #36: Found a better API

WouterJN, 2021-07-14:
While decompiling the Solargo app I found a better API that's refreshes the data more often and results in less errors

Get token:
`POST /api/Solargo/Login HTTP/1.1
Host: solargo.sems.com.cn
language: en
Content-Type: application/json
Content-Length: 74
{"sn": "_serial_of_goodwe_", "password": "_MD5_hash_of_password_"}
`

Geta data:
`GET /api/Solargo/GetRealdataOverview?language=%22en%22&sn=_serial_of_goodwe_ HTTP/1.1
Host: solargo.sems.com.cn
token: _token_
language: en
`

Are you willing to use this API for your HA Custom Component? If not then I will try to make my own component.

TimSoethout, 2021-07-15:
Wow, really cool! If it performs better, we could definitely use this, but
maybe keep the old URLs around?

I see this is a `.cn` domain, while the current `semsportal.com` seems to
redirect based on location to `.eu`. I don't know what the performance
difference will be?

I don't know when I will have time to look into this, so feel free to fork
the component and see it you can get it working. :)

PS there also seems to be some documentation in chinese:
http://solargo.sems.com.cn/help

WouterJN, 2021-07-16:
Time is also always an issue for me, hopefully I have some times after my holiday.

Meanwhile I made a temporally test in Node-red and the results are looking very promising, the data frequency is one minute instead of 5 minutes with the SEMS API.

Silv0rShad0w, 2022-09-26:
@WouterJN do you use the solargo API in Home Assitant??

closed 2026-09-19 by TimSoethout