TimSoethout/goodwe-sems-home-assistant #38: Attributes show unavailable

Robje007, 2021-08-05:
My added attributes are not working they are all showing as unavailable, I replaced $NAME with Goodwe (Thats how it is called in Home assistant)

This is my configuration.yaml:

`# Totaal verbruik kwh
sensor:
  - platform: template
    sensors:
      pv_outputpower:
        value_template: '{{ states.sensor.inverter_Goodwe.attributes.outputpower }}'
        unit_of_measurement: 'W'
        friendly_name: "PV Power output"
      pv_temperature:
        value_template: '{{ states.sensor.inverter_Goodwe.attributes.tempperature }}'
        unit_of_measurement: 'C'
        friendly_name: "PV Temperature"
      pv_eday:
        value_template: '{{ states.sensor.inverter_Goodwe.attributes.eday }}'
        unit_of_measurement: 'kWh'
        friendly_name: "PV energy day"
      pv_etotal:
        value_template: '{{ states.sensor.inverter_Goodwe.attributes.etotal }}'
        unit_of_measurement: 'kWh'
        friendly_name: "PV energy total"
      pv_iday:
        value_template: '{{ states.sensor.inverter_Goodwe.attributes.iday }}'
        unit_of_measurement: '€'
        friendly_name: "PV income day"
      pv_itotal:
        value_template: '{{ states.sensor.inverter_Goodwe.attributes.itotal }}'
        unit_of_measurement: '€'
        friendly_name: "PV income total"
      pv_excess:
        value_template: '{{ states.sensor.inverter_Goodwe.attributes.pmeter }}'
        unit_of_measurement: 'W'
        friendly_name: "PV spare"
      # battery soc
      pv_soc:
        value_template: '{{ states.sensor.inverter_Goodwe.attributes.soc }}'
        unit_of_measurement: '%'
        friendly_name: "Battery power"`

TimSoethout, 2021-08-06:
Hmm... That's odd. Anything in the log from SEMS?

You can also try the template on: http://hassio.lan:8123/developer-tools/template (replace your host name).
What does it show there?

Robje007, 2021-08-06:
Thanks for the suggestion
I used the (friendly) name instead of the entity-id, solved!

closed 2021-08-06 by Robje007