TimSoethout/goodwe-sems-home-assistant #41: Attributes not available anymore

werfpsa, 2021-08-06:
After the update of HA to 2021.8 and SEMS to 3.3 the attributes are not available anymore. One entity is available (sensor.inverter_none_energy) with the state for 'total generation' and the following attributes: 
state_class
last_reset
unit_of_measurement
friendly_name
device_class

Non of the other values are available anymore

TimSoethout, 2021-08-06:
There should be 2 entities, grouped under 1 device.
The first entity without "_energy" is the same as previously.

What do the error messages say?

werfpsa, 2021-08-06:
Tim,
I have deleted the without "_energy" entity as it was unavailable. There is no error message. 
API data received, but not available. See relevant log below:

[custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-08-06 17:58:07 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] API token not set (None) or new token requested (False), fetching
2021-08-06 17:58:07 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - Getting API token
2021-08-06 17:58:07 INFO (MainThread) 

2021-08-06 17:58:09 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Login Response: <Response [200]>
2021-08-06 17:58:09 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] SEMS - API Token received: {"uid": "28fb4b29-ce6a-4a2b-9173-f9e61bbebbc2", "timestamp": 1628265488993, "token": "958f1ef98efcfa1e5d0da3d051464372", "client": "ios", "version": "", "language": "en"}
2021-08-06 17:58:09 DEBUG (SyncWorker_3) [custom_components.sems.sems_api] Querying SEMS API for power station id: 29e01b10-d57a-4c6b-9408-42269bb7880e

2021-08-06 17:58:09 DEBUG (MainThread) [custom_components.sems.sensor] Finished fetching SEMS API data in 1.842 seconds

TimSoethout, 2021-08-06:
That is odd. Both should be available... The logging also look normal.

werfpsa, 2021-08-06:
Ok. But The recieven data is not…

werfpsa, 2021-08-07:
Tim,
I have done a complete new install and only one entity has been created, the one with '_energy'. So missing all the data

werfpsa, 2021-08-07:
Have downgraded HA to 2021.7.4 and sems 3.2 . Now I have the sensor.inverter_none back with all attributes.

TimSoethout, 2021-08-07:
Really odd. They should share data, so both should be there, but maybe the
coordinator works differently than I thought.

hans6910, 2021-08-10:
Same here. Reinstalled and configured, but get the same logs... Attributes come with the entity, but are not available in yaml for value states....

Logger: homeassistant.helpers.template
Source: helpers/template.py:1406
First occurred: 16:10:27 (21 occurrences)
Last logged: 16:10:56

Template variable warning: 'mappingproxy object' has no attribute 'Eday' when rendering '{{ states.sensor.inverter_gw_2500xs.attributes.Eday }}'
Template variable warning: 'mappingproxy object' has no attribute 'Etotal' when rendering '{{ states.sensor.inverter_gw_2500xs.attributes.Etotal }}'
Template variable warning: 'mappingproxy object' has no attribute 'Iday' when rendering '{{ states.sensor.inverter_gw_2500xs.attributes.Iday }}'
Template variable warning: 'mappingproxy object' has no attribute 'Itotal' when rendering '{{ states.sensor.inverter_gw_2500xs.attributes.Itotal }}'
Template variable warning: 'mappingproxy object' has no attribute 'Pmeter' when rendering '{{ states.sensor.inverter_gw_2500xs.attributes.Pmeter }}'

werfpsa, 2021-08-19:
Any idea when this issue will be resolved?

TimSoethout, 2021-08-19:
> Same here. Reinstalled and configured, but get the same logs... Attributes come with the entity, but are not available in yaml for value states....
>
> Logger: homeassistant.helpers.template
> Source: [...]

What do you get when trying `{{ states.sensor.inverter_zolder.attributes }}` on `http://hassio.lan:8123/developer-tools/template`?
For me all the attributes work when using a version with lower case: `{{ states.sensor.inverter_zolder.attributes.eday }}` (instead of `Eday`). Maybe changing how legacy attributes are handled in HA causes this.

werfpsa, 2021-08-20:
Tried 3.3 update again, but no attributes.
below log:
First occurred: 5:51:23 PM (2 occurrences)
Last logged: 5:51:23 PM

Error adding entities for domain sensor with platform sems
Error while setting up sems platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 457, in _async_add_entity
    device_info = entity.device_info
  File "/config/custom_components/sems/sensor.py", line 192, in device_info
    "model": self.extra_state_attributes["model_type"],
KeyError: 'model_type'

TimSoethout, 2021-08-20:
What does it show when enabling debug logs?
Maybe you have different attributes or something. The log should show what the API returns.

werfpsa, 2021-08-20:
2021-08-20 18:51:22 DEBUG (SyncWorker_2) [custom_components.sems.sems_api] SEMS - API Token received: {"uid": "28fb4b29-ce6a-4a2b-9173-f9e61bbebbc2", "timestamp": 1629478282715, "token": "b0b7629495a96983dba6ab6a84d4e348", "client": "ios", "version": "", "language": "en"}
2021-08-20 18:51:22 DEBUG (SyncWorker_2) [custom_components.sems.sems_api] Querying SEMS API for power station id: 29e01b10-d57a-4c6b-9408-42269bb7880e
2021-08-20 18:51:22 INFO (MainThread) 

[custom_components.sems.sensor] Found inverter attribute None 11500SSU13700112
2021-08-20 18:51:23 DEBUG (MainThread) [custom_components.sems.sensor] Finished fetching SEMS API data in 1.624 seconds
2021-08-20 18:51:23 INFO (SyncWorker_1) [homeassistant.loader] Loaded generic from homeassistant.components.generic
2021-08-20 18:51:23 DEBUG (MainThread) [custom_components.sems.sensor] Creating SemsSensor with id 11500SSU13700112
2021-08-20 18:51:23 DEBUG (MainThread) [custom_components.sems.sensor] Creating SemsStatisticsSensor with id 11500SSU13700112

[homeassistant.components.sensor] Error adding entities for domain sensor with platform sems
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 457, in _async_add_entity
    device_info = entity.device_info
  File "/config/custom_components/sems/sensor.py", line 192, in device_info
    "model": self.extra_state_attributes["model_type"],
KeyError: 'model_type'
2021-08-20 18:51:24 INFO (MainThread) 

after this below is repeated:
2021-08-20 18:53:24 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-08-20 18:53:24 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Querying SEMS API for power station id: 29e01b10-d57a-4c6b-9408-42269bb7880e
2021-08-20 18:53:24 DEBUG (MainThread) [custom_components.sems.sensor] Found inverter attribute None 11500SSU13700112
2021-08-20 18:53:24 DEBUG (MainThread) [custom_components.sems.sensor] Finished fetching SEMS API data in 0.180 seconds

TimSoethout, 2021-08-20:
It seems `model_type` is not available for you somehow:
Can you uncomment L184 in your `sensor.py`? https://github.com/TimSoethout/goodwe-sems-home-assistant/blob/master/custom_components/sems/sensor.py#L184
This enables more debug information.

Another way is to use your browser's developer tools when logging in to semsportal.com and capture the json for 
`https://www.semsportal.com/api/v2/PowerStation/GetMonitorDetailByPowerstationId` there to see if this attribute is different for you.

Which model do you have?

werfpsa, 2021-08-20:
the model field is empty '--' when I login to the goodwe portal.

2021-08-20 19:27:14 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry 29e01b10-d57a-4c6b-9408-42269bb7880e for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 293, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 90, in async_setup_entry
    return await component.async_setup_entry(entry)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 149, in async_setup_entry
    platform = await async_prepare_setup_platform(
  File "/usr/src/homeassistant/homeassistant/setup.py", line 329, in async_prepare_setup_platform
    platform = integration.get_platform(domain)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 498, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 503, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 846, in exec_module
  File "<frozen importlib._bootstrap_external>", line 983, in get_code
  File "<frozen importlib._bootstrap_external>", line 913, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/sems/sensor.py", line 185
    return {
            ^
IndentationError: unindent does not match any outer indentation level
2021

TimSoethout, 2021-08-20:
`_LOGGER` should start at the same identation as the next line `return`.
It seems that your model is different than mine/others somehow.

werfpsa, 2021-08-20:
[home-assistant.log](https://github.com/TimSoethout/goodwe-sems-home-assistant/files/7023313/home-assistant.log)

werfpsa, 2021-08-20:
The type is GW1500-SS (indicated on the inverter)

TimSoethout, 2021-08-20:
Unfortunately the log file does not contain the same error about the `model_type`, so the debug message showing all attributes is not printed.
However in the log there seem to be issues logging into the portal somehow. 
Are you by any chance using the SEMS iOS app at the same time? This might result in logging the component out.

werfpsa, 2021-08-20:
I did not use the SEMS iOS, but have another log
[home-assistant.log](https://github.com/TimSoethout/goodwe-sems-home-assistant/files/7023489/home-assistant.log)

TimSoethout, 2021-08-20:
`2021-08-20 20:27:16 DEBUG (MainThread) [custom_components.sems.sensor] self.device_state_attributes: None`

This seems to be the problem. Somehow the attributes are not filled for you...

werfpsa, 2021-08-20:
But Sith verstond 3.2 it is working. Has this been changed?

werfpsa, 2021-08-20:
Sorry
But with version 3.2 it is working

TimSoethout, 2021-08-20:
In 3.2 we used `extra_state_attributes` for attributes, and in 3.3 `device_state_attributes`.
`extra_state_attributes` is deprecated, and the new version seems to work fine for me and others...

werfpsa, 2021-08-20:
Must be the good we model. It’s from 2014-2015
What to do now?

TimSoethout, 2021-08-20:
Not sure. Somehow your attributes are empty, so not only the model. I thought that could never happen, and that the coordinator takes care of it.

Joepie007v2, 2021-08-20:
I also have the same inverter 1500SS and same problem that it is not working anymore. Have never used debugging in HA.

TimSoethout, 2021-08-20:
Then we can compare the json that the api returns. Can you show the result of "https://www.semsportal.com/api/v2/PowerStation/GetMonitorDetailByPowerstationId" when visiting semsportal.com? You need to go into you browser's Developer Tools to the "network" tab.

werfpsa, 2021-08-20:
[Getmonitordetail_result.txt](https://github.com/TimSoethout/goodwe-sems-home-assistant/files/7023684/Getmonitordetail_result.txt)

werfpsa, 2021-08-20:
Hope this is the good one.

TimSoethout, 2021-08-20:
Interesting. Both `name` and `model_type` are `null`. (in `inverter_full`).

I guess for both the code needs to be a bit more defensively programmed, a.k.a. being able to deal with `null`.

Joepie007v2, 2021-08-20:
I used Chrome, went to semsportal and logged in. Then used F12, went to Network and went to
https://www.semsportal.com/api/v2/PowerStation/GetMonitorDetailByPowerstationId 

I only get this:

{
  "language": "zh_CN",
  "function": null,
  "hasError": true,
  "msg": "ver is not fund",
  "code": "UnknownVer",
  "data": null,
  "components": {
    "para": null,
    "langVer": 111,
    "timeSpan": 0,
    "api": null,
    "msgSocketAdr": null
  }
}

TimSoethout, 2021-08-20:
Can you try this one?
https://github.com/TimSoethout/goodwe-sems-home-assistant/releases/tag/3.3.1-beta
In HA, HACS can be set to show beta versions.

TimSoethout, 2021-08-20:
> I used Chrome, went to semsportal and logged in. Then used F12, went to Network and went to
> https://www.semsportal.com/api/v2/PowerStation/GetMonitorDetailByPowerstationId
>
> I only get this:
>
> { [...]

I'm not sure. This seems to be a different issue.

werfpsa, 2021-08-20:
working. All the attributes are back again. Thanks!

Joepie007v2, 2021-08-20:
Can't find where to allow beta's in HACS. Want to try the beta

Joepie007v2, 2021-08-20:
Copied the files over the old files and after the reboot my attributes are working again. Unfortunately the sun isn't shining ;)
Many thanks and I hope this could be integrated in a next release. 👍

TimSoethout, 2021-08-21:
Fixed in release 3.3.1
https://github.com/TimSoethout/goodwe-sems-home-assistant/releases/tag/3.3.1

closed 2021-08-21 by TimSoethout