TimSoethout/goodwe-sems-home-assistant #42: Unable to fetch data from SEMS.

nazeg, 2021-08-06:
Hi, i just installed this plugin via HACS/Integration page and entered my username/password and powerstation id from semsportal link but i'm getting these errors:

Error while setting up sems platform for sensor
7:50:49 PM – (ERROR) Sensor
Unable to fetch data from SEMS.
7:50:49 PM – (ERROR) GoodWe SEMS PV API (custom integration)

These are the debug details:

2021-08-06 19:50:44 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Login Response: <Response [200]>
2021-08-06 19:50:44 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - API Token received: {"uid": "2749c6f2-****-****-****91112022e7a3", "timestamp": 1628268644590, "token": "b3a860ba**********3f89b0687f133d", "client": "ios", "version": "", "language": "en"}
2021-08-06 19:50:44 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Querying SEMS API for power station id: 669e30d9-715a-417f-a127-3b683849429a
2021-08-06 19:50:45 INFO (MainThread) [custom_components.hacs] <Integration hacs/integration> Registration completed
2021-08-06 19:50:45 INFO (MainThread) [custom_components.hacs] Restore started
2021-08-06 19:50:45 INFO (MainThread) [custom_components.hacs] Restore done
2021-08-06 19:50:45 INFO (MainThread) [custom_components.hacs] Setup task HacsSetupTask.SENSOR
2021-08-06 19:50:45 INFO (MainThread) [custom_components.hacs] Stage changed: HacsStage.WAITING
2021-08-06 19:50:45 INFO (MainThread) [custom_components.hacs] Setup complete, waiting for Home Assistant before startup tasks starts
2021-08-06 19:50:45 INFO (MainThread) [custom_components.hacs] HACS is enabled
2021-08-06 19:50:45 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.hacs
2021-08-06 19:50:46 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Query not successful (The authorization has expired, please login again.), retrying with new token, 2 retries remaining
2021-08-06 19:50:46 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-08-06 19:50:46 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] API token not set ({"uid": "2749c6f2-****-****-****-91112022e7a3", "timestamp": 1628268644590, "token": "b3a860ba**********3f89b0687f133d", "client": "ios", "version": "", "language": "en"}) or new token requested (True), fetching
2021-08-06 19:50:46 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - Getting API token
2021-08-06 19:50:48 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Login Response: <Response [200]>
2021-08-06 19:50:48 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - API Token received: {"uid": "2749c6f2-****-****-****-91112022e7a3", "timestamp": 1628268647918, "token": "d34e8915**********a206ad884b59c2", "client": "ios", "version": "", "language": "en"}
2021-08-06 19:50:48 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Querying SEMS API for power station id: 669e30d9-****-****-****-3b683849429a
2021-08-06 19:50:49 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Query not successful (The authorization has expired, please login again.), retrying with new token, 1 retries remaining
2021-08-06 19:50:49 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2021-08-06 19:50:49 INFO (SyncWorker_4) [custom_components.sems.sems_api] SEMS - Maximum token fetch tries reached, aborting for now
2021-08-06 19:50:49 ERROR (SyncWorker_4) [custom_components.sems.sems_api] Unable to fetch data from SEMS.
2021-08-06 19:50:49 DEBUG (MainThread) [custom_components.sems.sensor] Finished fetching SEMS API data in 6.743 seconds
2021-08-06 19:50:49 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up sems platform for sensor

What can be the problem?

TimSoethout, 2021-08-06:
Did this fix itself after a while?
This sometimes occurs if parallel queries to the api are done. Maybe not use the SEMS portal mobile app at the same time.

nazeg, 2021-08-06:
> Did this fix itself after a while?
> This sometimes occurs if parallel queries to the api are done. Maybe not use the SEMS portal mobile app at the same time?

Unfortunately no. I tried to install it and restart homeassistant several times, no fix. I don't use SEMS portal mobile app currently. I also logged off from sems website but it didn't help :(

TimSoethout, 2021-08-06:
Did you just install for the first time? Or did you update?
You can try to install the previous version 3.2.0.

nazeg, 2021-08-06:
I installed it a few months ago i think but i removed it because i was using other plugin (which was using local ip). I installed it yesterday again to use with Energy tab.

TimSoethout, 2021-08-06:
I see. Version 3.2.0 indeed does not support the Energy tab.
Just the `-energy` entity id should be enough for the Energy tab. Maybe this is enough for you?

nazeg, 2021-08-06:
I will try 3.2.0 now

nazeg, 2021-08-06:
I copied the 3.2.0 files manually but getting same errors :(

TimSoethout, 2021-08-07:
I don't know. It should work. Did you try increasing the poll rate? So more than 60s per call.
You can also login on semsportal.com?

nazeg, 2021-08-07:
I can login on semsportal without any problem. I just tried to increase poll rate but same error unfortunately :(

Login Response: <Response [200]>
Query not successful (The authorization has expired, please login again.), retrying with new token, 1 retries remaining
SEMS - Maximum token fetch tries reached, aborting for now

TimSoethout, 2021-11-24:
Closing this. Reopen if still an issue.

closed 2021-11-24 by TimSoethout

TimSoethout, 2021-12-20:
@nazeg , this might be fixed in 3.7.0-beta. I ran into a similar issue today.

The token authentication response also returns an `api` field with a url.
Now the next calls use this new api `url`, since the token should always work there (, at least that is my guess). It seems to work for me again now.