TimSoethout/goodwe-sems-home-assistant #44: Unable to prepare setup

HCookie, 2021-08-08:
Having some troubles setting up this integration.
After importing it through HACS, and then loading it into HomeAssistant and providing all the neccesary credentials, the following error was encounted.
` Unable to prepare setup for platform sems.sensor: Platform not found (cannot import name 'STATE_CLASS_MEASUREMENT' from 'homeassistant.components.sensor' (/usr/src/homeassistant/homeassistant/components/sensor/__init__.py)). `

There is no homeassistant folder at /usr/src/ so the error reason is clear, however, why this is happening is less so.
Homeassistant is running as `Home Assistant 2021.5.4` on a raspberry pi 4, in a docker-compose container.

TimSoethout, 2021-08-08:
I guess you need HA 2021.8 now.
Manifest needs to be updated still.

TimSoethout, 2021-08-08:
Or downgrade to 3.2.0.

HCookie, 2021-08-08:
Ahh why yes, that does make sense, thank you

closed 2021-08-08 by HCookie