TimSoethout/goodwe-sems-home-assistant #45: Inverter name shows 'None'

Calimerorulez, 2021-08-12:
Hi,

Thank you for creating this integration :)

I installed it on my HA, I've inserted my credentials, and retrieved the station ID via the URL, but HA shows two entities, 'Inverter None' and 'Inverter None Energy'. It does not show a correct name...

I'm also trying to integrate it with the new Energy menu item in the latest HA, but the net HA integration only lets me select the 'Inverter None Energy' which is the total of generated power over time, and I can't select the current generated power entity.

What am I doing wrong?

TimSoethout, 2021-08-12:
Hi, thanks. 

This is how it is supposed to work. The "None" name can be changed on semsportal.com or the app I think.

I think you're using it correctly. You only select the total kWh in the solar overview.
The power generated should come from a smart meter of other way of determining how much you use from/to the grid.
But maybe I'm missing something?

Calimerorulez, 2021-08-12:
Hi,

I then think I configured it as you intended. Will keep you posted. I'm not generating any solar power at the moment, it is getting dark in The Netherlands ;)

I will check tomorrow.

Calimerorulez, 2021-08-13:
Sorry for my confusion, it's working now... :)

closed 2021-08-13 by Calimerorulez