TimSoethout/goodwe-sems-home-assistant #46: Can not open config flow

johnolsthoorn, 2021-08-12:
Tim,
Bedankt voor je script,
Ik heb denk ik de files gecopieerd maar bij toepassing start sems niet en krijg ik de opmerking dat config flow niet te openen is.
Doe ik iets fout?

TimSoethout, 2021-08-12:
It should of course just work.
Is there anything in your log files? Are you using the latest Home Assistant version?

johnolsthoorn, 2021-08-12:
Bedankt voor je snelle reactie. Ik heb de regels gecompileerd vanaf je post en deze geplakt in de file. Ik merk alleen dat de 83 regels er 82 zijn ivm loze regel op 24 maar dat heeft mijn inziens geen effect omdat het quote betreft.

Bij starten geeft HA aan dat config flow niet geladen kan worden.

Kan ik de config flow file op een andere manier overzetten ipv copy en past van de 83 regels?

Outlook voor Android downloaden<https://aka.ms/ghei36>

johnolsthoorn, 2021-08-12:
NB ik gebruik laatste versie HA ( vandaag)

Outlook voor Android downloaden<https://aka.ms/ghei36>

johnolsthoorn, 2021-08-12:
> I'm not entirely getting how you configured it. The easiest install method is via HACS, see: https://github.com/TimSoethout/goodwe-sems-home-assistant#easiest-install-method-via-hacs
> The rest of the configuration in the readme is optional.
>
> Unfortunately there is no other configuration than the config flow (due to how HA works).
>
> Which exact messages are visible in Configuration > Logs?

Error handling request
21:34:23 – (FOUT) sems (aangepaste integratie)
bieniu/ha-gios - Repository is archived.
21:32:09 – (FOUT) HACS (aangepaste integratie) - bericht kwam voor het eerst om 21:32:09 en verschijnt 3 keer
Error with socket 48 (('2a02:a210:9600:9900:e27d:481e:ed9b:f4d6', 5353, 0, 0))): [Errno 101] Network unreachable
21:32:07 – (WAARSCHUWING) /usr/local/lib/python3.9/site-packages/zeroconf/_logger.py
Config entry 'ARRIS TG2492LG-85 Router' for upnp integration not ready yet: None; Retrying in background
21:31:56 – (WAARSCHUWING) config_entries.py
Waiting on integrations to complete setup: hacs, tuya, forecast_solar, sensor.dsmr, tts, mobile_app, shelly, yeelight
21:31:53 – (WAARSCHUWING) bootstrap.py
Setup failed for sems: unknown error
21:30:48 – (FOUT) sems (aangepaste integratie)

I get the message Config flow kan niet geladen worden

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows

johnolsthoorn, 2021-08-12:
Initially i saw the directory via installation via HACS but I manually copied the files according to the instruction on your link. I did  not checked if the files were identical.

So option to erase everything en redo via HACS?

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows

TimSoethout, 2021-08-12:
It should not matter how you installed it, but there could be a problem there.

The error message in your log
```Setup failed for sems: unknown error
21:30:48 – (FOUT) sems (aangepaste integratie)```
does not show everything. What do you see when clicking on it?

PS I think reinstalling via HACS should overwrite your manually installed version automatically (given that they were in the correct directory ;)).

johnolsthoorn, 2021-08-12:
I reinstalled via HACS and overwrite the files apparently correctly.
I could start
Config flow loaded.

In the log file i now see the following

Error while setting up sems platform for sensor
22:50:07 – (FOUT) Sensor - bericht kwam voor het eerst om 22:45:30 en verschijnt 2 keer
Unable to fetch data from SEMS.
22:50:07 – (FOUT) GoodWe SEMS API (aangepaste integratie) - bericht kwam voor het eerst om 22:45:28 en verschijnt 2 keer
bieniu/ha-gios - Repository is archived.
22:45:43 – (FOUT) HACS (aangepaste integratie) - bericht kwam voor het eerst om 22:45:43 en verschijnt 3 keer
Error with socket 37 (('2a02:a210:9600:9900:e27d:481e:ed9b:f4d6', 5353, 0, 0))): [Errno 101] Network unreachable
22:45:42 – (WAARSCHUWING) /usr/local/lib/python3.9/site-packages/zeroconf/_logger.py
Config entry 'ARRIS TG2492LG-85 Router' for upnp integration not ready yet: None; Retrying in background
22:45:37 – (WAARSCHUWING) config_entries.py - bericht kwam voor het eerst om 22:45:30 en verschijnt 2 keer

But i presume this is because the goodwe is now turned down because of night (by lack of light the inverter is switched off offcause)

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows

TimSoethout, 2021-08-13:
It should still work because it connects to the cloud service by GoodWe. 

`Error with socket 37 (('2a02:a210:9600:9900:e27d:481e:ed9b:f4d6', 5353, 0, 0))): [Errno 101] Network unreachable` seems to be a network issue. 
Maybe you can't connect to the internet via HA? Also seems to be an ipv6 address, which should work, but did not expect.

johnolsthoorn, 2021-08-13:
I have found IP adres and the ping reacted correctly.

Error i got was

Logger: homeassistant.components.sensor
Source: helpers/update_coordinator.py:161
Integration: Sensor (documentation, issues)
First occurred: 15:06:52 (1 occurrences)
Last logged: 15:06:52

Error while setting up sems platform for sensor
Traceback (most recent call last):
  File "/config/custom_components/sems/sensor.py", line 61, in async_update_data
    raise UpdateFailed(
homeassistant.helpers.update_coordinator.UpdateFailed: Error communicating with API, probably token could not be fetched, see debug logs

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 187, in _async_refresh
    self.data = await self._async_update_data()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 147, in _async_update_data
    return await self.update_method()
  File "/config/custom_components/sems/sensor.py", line 75, in async_update_data
    raise UpdateFailed(f"Error communicating with API: {err}")
homeassistant.helpers.update_coordinator.UpdateFailed: Error communicating with API: Error communicating with API, probably token could not be fetched, see debug logs

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/sems/sensor.py", line 96, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 161, in async_config_entry_first_refresh
    raise ex
homeassistant.exceptions.ConfigEntryNotReady: Error communicating with API: Error communicating with API, probably token could not be fetched, see debug logs

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows

johnolsthoorn, 2021-08-13:
I also saw this remark regarding wrong command

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:89
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 15:06:44 (1 occurrences)
Last logged: 15:06:44

[2968604648] Received invalid command: energy/info

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows

TimSoethout, 2021-08-13:
Sorry, that does not ring a bell. What do the debug logs show?
To see debug logs, add the last line in the correct place in your `configuration.yaml`:
```logger:
  default: info
  logs:
    custom_components.sems: debug
```

johnolsthoorn, 2021-08-13:
Ik krijg volgende info:

avdeevsv91/ha_generic_hygrostat - Repository is archived.
16:41:03 – (FOUT) HACS (aangepaste integratie) - bericht kwam voor het eerst om 16:41:03 en verschijnt 3 keer
Error with socket 43 (('2a02:a210:9600:9900:e27d:481e:ed9b:f4d6', 5353, 0, 0))): [Errno 101] Network unreachable
16:41:02 – (WAARSCHUWING) /usr/local/lib/python3.9/site-packages/zeroconf/_logger.py
Error while setting up sems platform for sensor
16:41:00 – (FOUT) Sensor
Unable to fetch data from SEMS.
16:41:00 – (FOUT) GoodWe SEMS API (aangepaste integratie)
Setup of sensor platform sems is taking over 10 seconds.
16:40:48 – (WAARSCHUWING) Sensor
Config entry 'ARRIS TG2492LG-85 Router' for upnp integration not ready yet: None; Retrying in background
16:40:42 – (WAARSCHUWING) config_entries.py
Platform error: sensor - Integration 'goodwe' not found.
16:40:35 – (FOUT) config.py

Details zijn:

2021-08-13 16:40:32 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration sems which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-08-13 16:40:32 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration omnik which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-08-13 16:40:32 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-08-13 16:40:33 INFO (MainThread) [homeassistant.setup] Setup of domain logger took 0.0 seconds
2021-08-13 16:40:33 INFO (MainThread) [homeassistant.setup] Setting up http
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain http took 0.5 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain recorder took 0.6 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up system_log
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain system_log took 0.0 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 1: {'frontend', 'api', 'lovelace', 'onboarding', 'person', 'image', 'http', 'system_log', 'device_automation', 'cloud', 'hassio', 'search', 'analytics', 'websocket_api', 'config', 'auth', 'webhook'}
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up api
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain api took 0.0 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up lovelace
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up device_automation
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain device_automation took 0.0 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up hassio
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up websocket_api
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain websocket_api took 0.0 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up config
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up auth
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain auth took 0.0 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up webhook
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain webhook took 0.0 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up image
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain config took 0.3 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up search
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain search took 0.0 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up analytics
2021-08-13 16:40:34 INFO (SyncWorker_3) [homeassistant.loader] Loaded google_assistant from homeassistant.components.google_assistant
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain lovelace took 0.4 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain analytics took 0.0 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain image took 0.0 seconds
2021-08-13 16:40:34 INFO (SyncWorker_4) [homeassistant.loader] Loaded alexa from homeassistant.components.alexa
2021-08-13 16:40:34 INFO (SyncWorker_1) [homeassistant.loader] Loaded camera from homeassistant.components.camera
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up person
2021-08-13 16:40:34 INFO (SyncWorker_3) [homeassistant.loader] Loaded media_player from homeassistant.components.media_player
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up cloud
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain person took 0.2 seconds
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setting up onboarding
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.helpers.storage] Migrating onboarding storage from 3 to 4
2021-08-13 16:40:34 INFO (MainThread) [homeassistant.setup] Setup of domain onboarding took 0.0 seconds
[... cut ...]
2021-08-13 16:40:48 INFO (MainThread) [zigpy_znp.zigbee.application] PAN ID: 0x4AA4
2021-08-13 16:40:48 INFO (MainThread) [zigpy_znp.zigbee.application] Extended PAN ID: 2e:db:03:aa:82:3b:bb:f4
2021-08-13 16:40:48 INFO (MainThread) [zigpy_znp.zigbee.application] Device IEEE: 00:12:4b:00:1c:d4:05:38
2021-08-13 16:40:48 INFO (MainThread) [zigpy_znp.zigbee.application] Device NWK: 0x0000
2021-08-13 16:40:48 INFO (SyncWorker_7) [homeassistant.loader] Loaded climate from homeassistant.components.climate
2021-08

TimSoethout, 2021-08-13:
The problem seems to be in:
`2021-08-13 16:40:38 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - Getting API token`.
It never finishes.
I'm not sure why though...

TimSoethout, 2021-08-21:
You can try chaning L15 in `sems_api.py`: `_RequestTimeout = 30  # seconds` to a longer time-out.

TimSoethout, 2021-11-24:
Closed for now. Please reopen if stillan issue.

closed 2021-11-24 by TimSoethout