TimSoethout/goodwe-sems-home-assistant #48: No longer working in HA after recent update

mareklab, 2021-08-24:
After recent update via HACS the integration stopped working with Home Assistant.

It still communicates with my inverter :

`2021-08-24 13:43:37 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Connecting to DT family inverter at 192.168.123.10:8899
2021-08-24 13:43:37 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Send: '7f03753100280409'
2021-08-24 13:43:38 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Received: 'aa557f0350000000000001353830303044545332303957303134324757384b2d4454202020ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff000003f203f2000802d800310001000004e4'
`

But then there is an error directly in HA:

`2021-08-24 13:43:48 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/goodwe/sensor.py", line 157, in async_refresh
    self.entity.update_value(inverter_response)
  File "/config/custom_components/goodwe/sensor.py", line 196, in update_value
    self.async_schedule_update_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 621, in async_schedule_update_ha_state
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 464, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 500, in _async_write_ha_state
    attr.update(self.state_attributes or {})
ValueError: dictionary update sequence element #0 has length 77; 2 is required`

I guess the problem is here: 

**ValueError: dictionary update sequence element #0 has length 77; 2 is required**

TimSoethout, 2021-08-24:
Hi, this seems to be for a different component?
This is the GoodWe sems plugin using the api, with debug name: `custom_components.sems`, not `custom_components.goodwe.goodwe_inverter`.

mareklab, 2021-08-24:
Good point, thanks! For a strange reason, SEMS directed me to your repository...

closed 2021-08-24 by mareklab