TimSoethout/goodwe-sems-home-assistant #49: Error " unexpected state class "  for sensor since homeasstant 2021.9

tabnul, 2021-09-03:
Since last update of home assistant (2021.9) , things seem to have changed for energy sensors.
I get below error for the SEMS sensor (also for all other energy logging sensors)

Unexpected state class
The following entities do not have the expected state class 'total_increasing'

[image]

gvssr, 2021-09-03:
Getting the same error on 2021.9.2

TimSoethout, 2021-09-04:
That is odd. I scanned the new documentation and `measure` sensor should not need this new property.
I will fix it when I have time to upgrade my HA version, which might take a while.

Jozziej, 2021-09-05:
> That is odd. I scanned the new documentation and `measure` sensor should not need this new property. I will fix it when I have time to upgrade my HA version, which might take a while.
> […](#)

Hi Tim,

I see you just commited a change, I also manually did the same manually on my end and it works.

Below import can also be removed now if i'm not mistaken
"from homeassistant.util.dt import utc_from_timestamp"

TimSoethout, 2021-09-05:
Eventually yes, but the `last_reset` is still there. I guess we can remove it too.

TimSoethout, 2021-09-05:
Removed it now in a new commit. Feel free to try the `master` branch.

Jozziej, 2021-09-05:
Ok yes, I also tried it on my end without the last reset, since it's already deprecated now.

TimSoethout, 2021-09-05:
Released in 3.4.0

closed 2021-09-05 by TimSoethout