TimSoethout/goodwe-sems-home-assistant #50: warning with 2021.9.2 : Setting last_reset is deprecated and will be unsupported from 2021.11

joostvkempen, 2021-09-03:
I've got a warning in the logging when starting Home Assistant 2021.9.2:

Entity sensor.inverter_goodwe3000_energy (<class 'custom_components.sems.sensor.SemsStatisticsSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11.

I think this is the change needed: https://developers.home-assistant.io/blog/2021/08/16/state_class_total/

related to #49

TimSoethout, 2021-09-05:
Thanks.

Feel free to try the `master` branch. I will promote it to a release if it works.

I don't see any warnings or errors anymore and Energy seems to work.

TheGabeMan, 2021-09-05:
Had the same issue, after applying your proposed change to my configuration, it worked.
That is change: 3bcef2eddedf7cbe3eb6369427044b11c320d89d 
While typing this is see you applied another change, "remove last reset", which I haven't applied yet.

TimSoethout, 2021-09-05:
Released in 3.4.0

closed 2021-09-05 by TimSoethout