TimSoethout/goodwe-sems-home-assistant #52: Any hope for integrating Home Kit?

darek-margas, 2021-09-10:
I've added some code to version 1.0.0 to read my Home Kit data. Still keep this version installed even though I migrated monitoring from Home Kit onto Iotawatt. Looked at your update yesterday but still only inverter. 
Would be nice to see Home Kit too.

I can paste here entire json downloaded from account having Home Kit, probably can fork 1.0.0. and paste my messy code too.

TimSoethout, 2021-09-11:
Hi, I'm not sure what you want? Home Kit is from Apple, right? I guess that integrations is handled by Home Assistant and not this individual plugin?

darek-margas, 2021-09-11:
https://www.goodwe.com.au/monitoring-system/homekit-299_7.asp
Readable from the same SEMS API. Does report consumption, etc.

Logzy72, 2021-10-17:
I've also got the Goodwe Homekit and would like to see the house consumption etc in HA. 
Did you fork this with additional code. If so, where can I find this.? 
Specifically it would be great to see consumption, solar gererated and returned to grid.

TimSoethout, 2021-10-17:
Hi, this should already work for a while. Did you upgrade the integration?

Logzy72, 2021-10-17:
Hi.
Im running the latest integration but the only 2 sensor it gives me is
current production and total production.

sensor.inverter_55000msu218w0020_energy

sensor.inverter_55000msu218w0020

darek-margas, 2021-10-17:
> 
> 
> I've also got the Goodwe Homekit and would like to see the house consumption etc in HA. Did you fork this with additional code. If so, where can I find this.? Specifically it would be great to see consumption, solar gererated and returned to grid.

I did fork it recently but did not write anything yet. I've made working code from version 1 just never made it public as it is mess with no conditions, etc, just assuming homekit is present.

With this newer version I had a few thoughts but got stuck between adding simplest code or, say, trying to discover its presence, matching with inverter, etc.

If someone else has more time and experience I am happy to provide json with description what is what and for what exactly.
SEMS API, however, is not perfect. HK provides another instance of generation plus grid clamp reading. Generation slightly differs compared to inverter reading. For grid clamp API returns just unsigned power and then grid status as 1 or -1 which then can be used to make signed power or two separate values when one is always 0. With accumulated generation/usage values for some reason API winds back values after midnight. Not much but still like 1h usage on average. I didn't find its cause and as this confuses energy dashboard I used Riemann sum calculated from power.

Ads5555, 2021-10-31:
Created this. Let me know what you think, my first attempt at python and HA integration. https://github.com/TimSoethout/goodwe-sems-home-assistant/pull/58

closed 2021-10-31 by TimSoethout