TimSoethout/goodwe-sems-home-assistant #53: Using Goodwe sems sensor data to populate new HA Energy tab

JoLambinon, 2021-09-12:
I have tried to use the available entities that are recognised by the new HA energy function but all sensors/entities are not available/recognised?

JoLambinon, 2021-09-12:
Apologies to everyone I am new to Github HA.  Just realised there is already a lot of information about this issue.
Thank you for the fantastic work that you all do !!

closed 2021-09-12 by JoLambinon

calebvisage, 2022-01-31:
Hi @JoLambinon, I'm facing the same issue. Could you please elaborate on what you did to resolve this?

https://xkcd.com/979/

TimSoethout, 2022-02-01:
@calebvisage, My guess is that you need to upgrade to the latest SEMS component version (via HACS), or that you have to wait for a couple hours before HA populates the statistics.
Another guess is that he found some info in other issues, such as #37 ?

calebvisage, 2022-02-01:
@TimSoethout I can get them recognized by adding the `device_class` attribute but now I cannot add a `state_class`.

calebvisage, 2022-02-01:
@TimSoethout I've taken a look at that issue but did not found it helpful, although I will take a look again in case I missed anything.

calebvisage, 2022-02-01:
@TimSoethout Also I have installed it manually and assume it's the latest version.

TimSoethout, 2022-02-02:
That should work indeed. Also running the latest HA? They changed the API for `device_class`/`state_class` in the last half year...

calebvisage, 2022-02-02:
Yes. It was my template that was wrong. I got it working :) Just one issue... the entities become unavailable quite often and values seem to be lagging behind on the Energy dashboard (or are just plain wrong). Is this normal? (Also, should I create a new issue for this since idk if it's "okay" to continue a thread on a closed issue. Thanks for helping btw and thank you for this integration.

TimSoethout, 2022-02-03:
Great that it works now.

Not sure about the wrong values. It's not unavailable for me... Although I still have to upgrade to latest HA version myself.
Feel free to open a new issue indeed. :) Maybe add some screenshots.