TimSoethout/goodwe-sems-home-assistant #54: Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11

MPowerData, 2021-09-18:
Hi,

Since the latest upgrade of Home Assistant below warning is showing up in the log.

WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.inverter_gw3600d_ns_energy (<class 'custom_components.sems.sensor.SemsStatisticsSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.

Is this something I can fix?

Or does the integration need an update?

Thanks for your help.

TimSoethout, 2021-09-19:
The latest version should not have `last_reset` anymore. Did you update the component?

MPowerData, 2021-09-19:
Hmmm, how do I run the update?

TimSoethout, 2021-09-19:
Via HACS?
Or copy the files from this repo to your `custom_components` directory.

MPowerData, 2021-09-19:
Thanks, that worked!

closed 2021-09-19 by MPowerData