TimSoethout/goodwe-sems-home-assistant #55: Nastavení hloubky výboje z portálu SEMS

sotoniak, 2021-09-23:
Hi I search solution for use case, when I charge my car at night and I don't want discharge home battery. I need thru Home Assistant disconnect battery. GoodWe doesn't have this function (without change inverter mode and restart), but I think that workaround is setting DoD to 1%. DoD is possible setting only in Android application Sems Portal (no in the web interface, or iOS). Has a API available this function?

Thank you Tom

TimSoethout, 2021-09-24:
Tbh I have no clue. If you can change this setting via the app or website, it can probably also be done via the API.

spockz, 2021-09-24:
Setting the DoD to 1% means also not serving the rest of your house's power needs from the battery. Are you sure that is what you want?

I'm unsure whether the BMS exposes this option but basically you want the battery output to be `meter usage-car charger consumption`.

What are your reasons for wanting to prevent charging the car with the battery? Do you have heavily fluctuating tariffs and want to use the battery to flatten peaks during high tariff?

sotoniak, 2021-09-24:
Hi, I have two reasons.
1. My car have 80kWh battery and my hause only 16 kWh. I want to avoid unnecessary battery discharge cycles.
2. Yes, I have two tariffs. The tariff for a car from 8 pm to 6 am and is 3 times cheaper than the normal price.

When I think about it, it will probably be easier to connect the WallBox via two contactors. Before (Charging from GRID only, low power) or after (Charging from the sun and batteries, high power) GoodWe SMART Meter. I will control the contactors via a relay and Home Assistant.

Thanks, I'm clear on that :-)

closed 2021-09-24 by sotoniak

rinners, 2022-04-26:
> Hi I search solution for use case, when I charge my car at night and I don't want discharge home battery. I need thru Home Assistant disconnect battery. GoodWe doesn't have this function (without [...]

Hi, did you ever figure out a way to do this? I see there has been an integration released since you asked, but it's mostly just polling, however I see that there is a drop down to be able to change the operation mode, so not sure why the discharge depth wasn't included.

https://www.home-assistant.io/integrations/goodwe/

yelonek, 2023-11-14:
> I see there has been an integration released since you asked which includes a depth of discharge entity.
> 
> https://www.home-assistant.io/integrations/goodwe/

Are you positive that this integration allows settings DoD? Or just displays it?

Where is the documentation for this plugin that would show this?

BTW I know it's old, but it is possible to switch to Economic Mode and set a "Charge time" for battery with low % of power. That will prevent the battery from discharging during that period. That also means that your home won't be supplied with power from the battery.

rinners, 2023-11-15:
> Are you positive that this integration allows settings DoD? Or just displays it?
>
> Where is the documentation for this plugin that would show this?
>
> BTW I know it's old, but it is possible to switch [...]

It definitely allows me to change DoD setting with my ES model. It does allow changing of modes, however with certain firmware versions this doesn't seem to do anything. I've got an older firmware version that allowes it but I still cant set up the Economic Mode settings with it. Best I can do is change it to Backup Mode. Just means being a little more cautious with DoD settings and paying more attention to the weather, but that is kind of what I want anyway. Do you not have panels?

yelonek, 2023-11-15:
@rinners Hi! I do have panels, but winter is coming and cloudy, so not much of a help right now. :)
Cool, I need to install Home Assistant then. I have GoodWe integration via Modbus in Loxone, but it's read-only. 
It would be awesome to be able to change anything. I don't understand however that it is not possible to change charge/discharge times through SEMS.