TimSoethout/goodwe-sems-home-assistant #59: Goodwe Homekit

Logzy72, 2021-11-02:
Great work on the addition of the Homekit sensor.
The data from this sensor in HA I have worked out from the data and comparing to my SEMS screen is the "consumption".

This sensor however is not available in the drop down box when trying to configure HA Energy.
This is also the same for trying to add any other Goodwe sensor to the HA Energy, they are not available to select to "add solar production", "add consumption" or "add return".

The Goodwe homekit has 2 CT's one reads solar production and the other reads the "feed from" or "return to" the grid.
From here consumption should just be a calculation of solar production, plus minus feed to or from the grid.
This is obviously how the SEMS portal calculates it.
I think we should be seeing 3 Homekit sensors. (From/to grid, consumption, solar production)

Not being able to add the Homekit sensor to HA is aparrantly due to the setup/coding of the sensor;
This is what HA says:
You’re configuring a statistic but you couldn’t find your source in the dropdown? That’s caused by a bug in the integration providing the entity. Integrations need to configure their entities correctly so Home Assistant knows that we need to track statistics for it and how.

Open an issue with the author of the integration and link them to https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics.

TimSoethout, 2021-11-02:
Yes indeed. For the statistics to work, you need a separate sensor with the correct `state_class`.

bitcrumb, 2021-11-10:
The README mentions how to create your own additional sensors from the state of the main sensor exposed by this integration. However, isn't it a better idea to expose all values in the state as separate sensor by default? This would allow the integration in add the proper `state_class`? Since I am not sure this is possible using the template integration as explained in the README.

An alternative Goodwe integration (which works via UDP, see: https://github.com/mletenay/home-assistant-goodwe-inverter) does exactly this.

gemmaker, 2022-05-18:
on core-2022.5.4
HACS 1.24.5
RaspberryPi4
GoodWe SEMS API Beta 3.7.0

Firstly- great work @TimSoethout . I was unable to use any of the local integrations with my Goodwe inverter because it upset the SEMS uploads (even if I set it to poll every 3 or 5 mins!) and none seem compatible with the GoodWe homekit.

I have a GoodWe inverter (GW10KAU-DT), plus GoodWe homekit (for three-phase power).
I have got the SEMS beta version installed which yeilds two devices (the homekit and the inverter). The Homekit device has three entities.  Two of which I can put straight into the HA Energy page:

- sensor.homekit_XXX_ import
- sensor.homekit_XXX_export

both in kWh of what should be only increasing values of my total grid consumption (eg buying), and grid export (selling) respectively. 
However every night, just after midnight I'm recording backwards values, followed by a 2:30am recovery for both the cumulative inport and export:
[image]

This affects the HA Energy page.
Has anyone got a work around or an explanation for the behaviour?

philipbrennan, 2022-08-17:
```
template: 
  - sensor:
      - name: "PV Front"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.pv }}'
        unit_of_measurement: 'W'
        state_class: "measurement"
  - sensor:
      - name: "Grid"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.grid }}'
        unit_of_measurement: 'W'
        state_class: "measurement"
  - sensor:
      - name: "House Consumption"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.load }}'
        unit_of_measurement: 'W'
        state_class: "measurement"
  - sensor:
      - name: "PV Status"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.PowerFlowDirection }}'
        unit_of_measurement: 'W'
  - sensor:
      - name: "PV generation total"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.all_time_generation }}'
        unit_of_measurement: 'kWh'
        state_class: "total_increasing"
        device_class: "energy"
  - sensor:
      - name: "PV export total"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.Totals_sell }}'
        unit_of_measurement: 'kWh'
        state_class: "total_increasing"
        device_class: "energy"
  - sensor:
      - name: "PV import total"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.Totals_buy }}'
        unit_of_measurement: 'kWh'
        state_class: "total_increasing"
        device_class: "energy"
  - sensor:
      - name: "PV generation today"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.Charts_sum }}'
        unit_of_measurement: 'kWh'
        state_class: "measurement"
        device_class: "energy"
  - sensor:
      - name: "PV export today"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.Charts_sell }}'
        unit_of_measurement: 'kWh'
        state_class: "measurement"
        device_class: "energy"
  - sensor:
      - name: "PV import today"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.Charts_buy }}'
        unit_of_measurement: 'kWh'
        state_class: "measurement"
        device_class: "energy"
  - sensor:
      - name: "PV Self Use Today"
        state: '{{ states.sensor.solar_pv_91000hku21600330.attributes.Charts_selfUseOfPv   }}'
        unit_of_measurement: 'kWh'
        state_class: "measurement"
        device_class: "energy"
```

Denifia, 2022-10-10:
This seems closely related to #73 and #71, all of which might be addressed with the latest release.

gemmaker, 2022-10-11:
> This seems closely related to #73 and #71, all of which might be addressed with the latest release.

Let you know in 12 hours or so! 🙏

Denifia, 2022-10-11:
> Let you know in 12 hours or so! 🙏

I hope it fixes your issue!

I didn't think the PR would be merged so quickly or really at all so I didn't get a chance to mention that when switching from Totals to Charts data you'll see a new kind of spike in your graph - but it's only once off. 

Luckily most people with this issue have broken graphs already and the next day should have a fixed graph.

For my own sake I'm also writing a tool to repopulate all available sems data to give me clean historical graphs.

gemmaker, 2022-10-12:
> I hope it fixes your issue!
>
> I didn't think the PR would be merged so quickly or really at all so I didn't get a chance to mention that when switching from Totals to Charts data you'll see a new [...]

Thanks so much @Denifia everything is working! Yes I had the one-off spike at midnight (reverse to the usual one) and it is good since.
I'd love access to your tool to clean up my old data as well, if that is possible.
Thanks again for your effort.
Now, the 🌞 just has to shine!

Denifia, 2022-10-12:
@gemmaker glad the fix worked for you! 

I'll share my cleanup tool when it's working for me but I don't get a lot of time to work on it so it could be a few weeks away.

Denifia, 2022-10-18:
@gemmaker I've put up my first cut of a Sems HomeKit importer at https://github.com/Denifia/goodwe-sems-history-home-assistant but in some weird turn of fate that has a bug where recalculated days that were _not_ imported have the daily data spike.

I've run out of time to work on it but hopefully someone else in the community will be able to contribute.

TimSoethout, 2026-02-06:
Closing, please open if still relevant. :)

closed 2026-02-06 by TimSoethout