TimSoethout/goodwe-sems-home-assistant #63: Negative power

TonKro, 2021-12-30:
Lately while the solar is not active I get high kwh peaks back. Some times later these peaks are "equalized" with a negative peak:

[image]

[image]

What could cause this and how can I solve it?

TimSoethout, 2022-01-01:
That is really odd... The energy should be an always increasing unit. We take the `data["etotal"]` from the API, so that seems to return different values (older? or of the other inverter?) for you somehow.

TonKro, 2022-01-03:
There are not more then the 3 showed inverters. I do not get it either, but somehow as the problem did appear it did disappear. For now let's close the issue, I will keep a close eye on it.

Thanks a lot.

TimSoethout, 2022-01-03:
If it happens again you can try turning on debug logging. It should log the JSON from the API as well. See the readme on how to do this.
Good luck!

closed 2022-01-03 by TimSoethout