TimSoethout/goodwe-sems-home-assistant #64: Error adding entities for domain sensor with platform sems

Almendrita55, 2022-01-23:
Hi!

I'm a newbie on HA and trying to add integrations. I'm having problems with sems integration. Here is the log:

Logger: homeassistant.components.sensor
Source: custom_components/sems/sensor.py:166
Integration: Sensor (documentation, issues)
First occurred: 17:08:51 (2 occurrences)
Last logged: 17:08:51

Error adding entities for domain sensor with platform sems
Error while setting up sems platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 447, in _async_add_entity
    suggested_object_id = entity.name  # type: ignore[unreachable]
  File "/config/custom_components/sems/sensor.py", line 166, in name
    return f"Inverter {self.coordinator.data[self.sn]['name']}"
KeyError: 'name'

I don't know what's happening. Can anyone help me?

Thx in advanced

TimSoethout, 2022-02-01:
I'm not sure to be honest.
It could be that that your API response does not contain `name`.
Does your inverter have a name when manually loging in to www.semsportal.com?

Or it really is a bug where the `coordination.data` is not properly filled yet somehow.

Almendrita55, 2022-02-05:
Hi! thx for ur response.
 
Yes I have a name when manually log in to semsportal.com. May I have to introduce information on the code? What kind of information and where?

Thxs so much for your help.

TimSoethout, 2022-02-06:
You need to connect to the API using the configaration flow. I guess you did that. Sensors (with name) are then automatically added for all receivers.

rvilla87, 2022-02-21:
Hello, I have exactly the same problem:

```
Error adding entities for domain sensor with platform sems
Error while setting up sems platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 447, in _async_add_entity
    suggested_object_id = entity.name  # type: ignore[unreachable]
  File "/config/custom_components/sems/sensor.py", line 166, in name
    return f"Inverter {self.coordinator.data[self.sn]['name']}"
KeyError: 'name'
```

In my case, I was using `goodwe-sems-home-assistant-3.6.0-beta` since December with no problem but since 2-3 days ago I am not receiving 1 of all 5 sensors: `HomeKit 91000HKUxxxxxxx` is showing 0 W.

I tried with 3.4.0 version but I didn't remember that this version doesn't show the HomeKit information. I have also tried with 3.7.0-beta and still I get the same error.

Is there any way I can debug the python code that get information from semsportal without requering homeassistant? I have tried a simple test but I get the error `No module named 'homeassistant'`.

Thanks!

darek-margas, 2022-02-22:
I can't see now as it is night here but I fetch the same json to feed zabbix and I see that today from around noon in GMT+11 load got lost. It did happen previously so it is either modification or simply portal failure.

darek-margas, 2022-02-22:
Now zabbix is back and HA has data. I use 3.7.0-beta.
[image]
[image]

rvilla87, 2022-02-23:
Ok, then as it seems it's probably a problem trying to get the info of my plant in semsportal. In semsportal (via web and via app) it shows me the load electricity with no problem, but this integration stopped to work for me since February 19 at 18:18h (CET). It's strange because the other sensors work correctly.

This is my first time debugging a home assistant integration, I followed the [developement setup](https://github.com/TimSoethout/goodwe-sems-home-assistant#development-setup) but I'm not sure my dev environment of home assistant is working ok (it gets stuck on "`[homeassistant.components.zeroconf] Starting Zeroconf broadcast`".

Do you have any snippet in order to test the python code that gets info from semsportal without needing of home assistant libraries? Once I tested and fix the problem I could test it on my raspberry home assistant. For example as [this other project](https://pypi.org/project/goodwe/) that works for the inverter in local mode via UDP.

I thought of testing `sensor.py` removing all homeassistant dependencies but before spending time doing this I prefer to make sure this is a correct way to try to fix my problem with the integration.

Thanks in advance!

darek-margas, 2022-02-23:
So, are you stuck on configuring or it is set up but has no sensor data? I use curl to fetch sems API json. It is tricky as token has to be taken first. I think I published my zabbix stuff somewhere and it has bash script doing it but I don't remember where :P I'll try to put this stuff into my github account so that you would be able to download bash script frem there.

TimSoethout, 2022-02-23:
You can login to sems portal via the command line by doing the same steps as the component.
I use the Python `httpie` tool for this:

```
> http -v https://www.semsportal.com/api/v2/Common/CrossLogin account=XXXX pwd=XXXX token:'{"version":"3.1.1","client":"web","language":"en"}'
POST /api/v2/Common/CrossLogin HTTP/1.1
Accept: application/json, */*;q=0.5
Accept-Encoding: gzip, deflate
Connection: keep-alive
Content-Length: 54
Content-Type: application/json
Host: www.semsportal.com
User-Agent: HTTPie/2.6.0
token: {"version":"3.1.1","client":"web","language":"en"}

{
    "account": "XXXX",
    "pwd": "XXXX"
}

HTTP/1.1 200 OK
Access-Control-Allow-Headers: Content-Type
Access-Control-Allow-Methods: GET, POST, PUT, DELETE, OPTIONS
Access-Control-Allow-Origin: *
Cache-Control: no-cache
Connection: keep-alive
Content-Length: 532
Content-Type: application/json; charset=utf-8
Date: Wed, 23 Feb 2022 08:31:19 GMT
Expires: -1
Pragma: no-cache
Server: nginx
X-AspNet-Version: 4.0.30319
X-Powered-By: ASP.NET

{
    "api": "https://eu.semsportal.com/api/",
    "code": 0,
    "components": {
        "api": "http://eu.semsportal.com:82/api/Auth/GetTokenV2",
        "langVer": 129,
        "msgSocketAdr": "https://eu-xxzx.semsportal.com",
        "para": null,
        "timeSpan": 0
    },
    "data": {
        "client": "web",
        "language": "en",
        "timestamp": 1645605079188,
        "token": "4a22030XXXXc46d676",
        "uid": "XXXX-12345",
        "version": "3.1.1"
    },
    "hasError": false,
    "msg": "Successful"
}
```
Then use the content of `data` as token for the follow up requests:
```
> http -v https://eu.semsportal.com/api/v2/PowerStation/GetMonitorDetailByPowerstationId powerStationId=XXXX token:'{        "client": "ios",        "language": "en",        "timestamp": 1645605079188,        "token": "4a22030XXXXc46d676",        "uid": "XXXX-12345",        "version": "3.1.1"    }'

```

However at the moment it does not seem to work anymore...

Another option is to visit the website and collect the json in the network tab of your browsers developement tools.

rvilla87, 2022-02-24:
Thanks for the help!

I tested the commands that you said and I don't get the error `The authorization has expired, please log in again` if I set the same client in the request `GetMonitorDetailByPowerstationId` that I get in the token. So, If I get `web` client in the token, I set `web` in the request and it works correctly (I can get the data). If I set `ios` I get that error.

Yerterday I started to debug the `sems_api.py` in order to see why in my integration I see 0W in my home's electricity load and what a surprise to see that today at 4am my home assistant started to get my home's electricity load! I was sleeping, so I did not fix anything.

The thing is that in semsportal since forever I get correctly the data... well, the problem seems fixed, if the problem occurs again in the future I would debug again in order to see why it's failing to get the electricity load data.

Thanks!

TimSoethout, 2022-02-24:
Glad that it works again, but indeed really weird that it did not before...

closed 2022-02-24 by TimSoethout