TimSoethout/goodwe-sems-home-assistant #65: sensor.inverter_$NAME_energy is missing in HA energy dashboard

matskoeneman, 2022-02-23:
Dear Tim,

Thank you for all the work of this HACS integration. I'm running 3.7.0 beta, and the regular sensor is working fine with all the attributes. The sensor.inverter_$NAME_energy is reading the correct state corresponding to the 'eday' attribute however I cannot select it as Solar Production in the Energy Configuration.

The sensor has the following properties, which seems okay:

state_class: total_increasing
unit_of_measurement: kWh
device_class: energy

I can only guess for a possible solutions (by browsing the HA community) but not sure if this is the problem.
- adding 'unit_prefix: k' (this is often found in total_increasing energy sensors)
- adding 'round: 2' (now it has one decimal and in all other examples I see two, don't now if this will add a zero)
- handling 'unavailable' states
 
Cannot check this last one but what I found in the documentation is that this is used for unavailable values in the sensor:

availability: >
      {{ not 'unavailable' in 
             [ states('sensor.inverter_$NAME_energy') ] }}

TimSoethout, 2022-02-23:
Could it be that it takes a while before the energy sensor is available in the energy dashboard? I remember it takes ~2hours before statistics are available.

matskoeneman, 2022-02-23:
I have the integration running from 22 January now, with this long term statistic not working from the start. As a workaround I used:

platform: integration
  source: sensor.inverter_gw3000_xs
  name: inverter_total_production
  unit_prefix: k
  round: 2

Which seems to do the job but turns out to be too inaccurate with sunnier days. If I’m the only one with this problem I will troubleshoot some more. Removing the integration and add it again, or rebuilding / resetting some invalid database.

matskoeneman, 2022-02-25:
Nevermind, I overlooked a stupid mistake. In the configuration.yaml the 'recorder' was set to only include certain entity(_globs) I included the sensor.inverter_$name however the sensor.inverter_$name_energy is a separate one of course. So changing it the include entity_glob sensor.inverter* solved the problem.

closed 2022-02-25 by matskoeneman