TimSoethout/goodwe-sems-home-assistant #67: Goodwe SEMS API not available in integrations (HA-2022.4.6)

chalmers68, 2022-05-15:
Running home-assistant-2022.4.6 on Archlinux Arm RaspberryPi.
Installed Goodwe SEMS API manually by copying ```custom_components/sems/``` to ```custom_components/sems/``` in configuration directory (```/var/lib/hass```) and restarted Home Assistant.
After restart SEMS is not available in the integrations.

This is the only reference sems I can get in the log.
```
2022-05-15 22:51:42 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration sems which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
```
I also get these messages:
```
2022-05-15 22:56:59 WARNING (MainThread) [homeassistant.bootstrap] Setup timed out for bootstrap - moving forward
2022-05-15 22:57:04 ERROR (MainThread) [homeassistant] Error doing job: Task was destroyed but it is pending!
```

Relatively new to HomeAssistant. Not had to debug anything so far, so not sure how to get more detail about the problem.

TimSoethout, 2022-05-16:
The log message you mention from `homeassistant.loader` is normal behavior. It should show up when you try to add a new integration.

chalmers68, 2022-05-16:
I did that. But other than the Error and Warnings I listed there was not much else but the usual start-up info message.
Is there anything I should add to the debug logging config to shed some more light on what the issue is?

I saw an older issue with the same problem and an update to  HA 2021.6.5 fixed the problem.
I'm now on a newer version and although it is not the very latest, is is the latest official archlinux build available at the moment.
Is it possible another incompatibility has been introduced in this more recent HA 2022 version.
I have not tried installed via HACS yet. Unless HACS is a dependency, I'd rather not at the moment. I'd also rather understand why the manual install does not work if that is the case.

TimSoethout, 2022-05-16:
HA version`2019.9.3` and up should work.
https://github.com/TimSoethout/goodwe-sems-home-assistant/blob/master/hacs.json#L5

HACS is not a dependency, but just a nice way to install custom components and keep them updated. It should not be necessary for it to work.

chalmers68, 2022-05-17:
I get that it should work. But in my case it is not working. I'm running a stock archlinux install on a RaspberryPi. Adding this component is the only customisation I have made so far.
I'd really like to use this component, as my inverter is not supported by the Goodwe-inverter integration.

Some advise on how to generate useful debug information to diagnose the problem would be useful. I expect it has to be related to registration of the component in HA as it is not even appearing as an available integration. So that should direct what I should be looking for. Is there other components in HA I should enable debugging for?

TimSoethout, 2022-05-17:
I do see
```
2022-05-17 12:45:07 INFO (MainThread) [homeassistant.setup] Setting up sems
2022-05-17 12:45:07 INFO (MainThread) [homeassistant.setup] Setup of domain sems took 0.0 seconds
``` when rebooting my HA, but that could also be because it is loading my sensor.

HA probably supports more debug logging for loading components. Maybe you can try playing with HA debug settings: https://www.home-assistant.io/integrations/logger/

TimSoethout, 2022-05-17:
From the top of my head these things might be wrong:
- searching in the wrong place for adding integration: should be `Settings`>`Devices & Services`>bottom right `Add Integrations> search for `sems`

chalmers68, 2022-05-17:
I don't have "Devices & Services" in settings. It is in own section in the configuration. Integrations is the first panel there. "add Integrations" is on the bottom right. This is my search result.
[image]

I did look at enabling debug for other elements of HA. The problem is I don't know what the various logging elements in HA are. Or what their names might be. So I don't know what to put in.
I could try ```default: debug``` but that might be a bit excessive. I was hoping you would have a better idea on what to look for.

chalmers68, 2022-05-17:
OK this is weird. After enabling ```homeassistant.setup: debug``` and restarting I eventually got this in the log
```
2022-05-17 21:53:27 DEBUG (MainThread) [custom_components.sems.config_flow] SEMS - new config flow
```
and it now appears in the integration search.
I've restarted HA at least 3 or 4 times since installing the Sems component and this is the first time it has appeared.

I guess the problem has resolved itself somehow. I just don't know how.

closed 2022-05-17 by chalmers68

TimSoethout, 2022-05-17:
Yes, really odd indeed. Maybe it has something to do with delayed loading...
Glad it works now.