TimSoethout/goodwe-sems-home-assistant #68: Error communicating with API, probably token could not be fetched

ivo-toby, 2022-06-11:
The integration in home-assistant stopped working about 2/3 days ago. I tried to reinstall and another (visitor) account but still no success. 

The logs in HA say:
```
2022-06-11 16:51:27 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform sems is taking over 10 seconds.
2022-06-11 16:51:28 ERROR (MainThread) [frontend.js.latest.202206010] :0:0 ResizeObserver loop completed with undelivered notifications.
2022-06-11 16:51:58 ERROR (SyncWorker_2) [custom_components.sems.sems_api] Unable to fetch data from SEMS. 
2022-06-11 16:51:58 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up sems platform for sensor
Traceback (most recent call last):
  File "/config/custom_components/sems/sensor.py", line 60, in async_update_data
    raise UpdateFailed(
homeassistant.helpers.update_coordinator.UpdateFailed: Error communicating with API, probably token could not be fetched, see debug logs

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 191, in _async_refresh
    self.data = await self._async_update_data()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 150, in _async_update_data
    return await self.update_method()
  File "/config/custom_components/sems/sensor.py", line 74, in async_update_data
    raise UpdateFailed(f"Error communicating with API: {err}")
homeassistant.helpers.update_coordinator.UpdateFailed: Error communicating with API: Error communicating with API, probably token could not be fetched, see debug logs

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/sems/sensor.py", line 95, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 164, in async_config_entry_first_refresh
    raise ex
homeassistant.exceptions.ConfigEntryNotReady: Error communicating with API: Error communicating with API, probably token could not be fetched, see debug logs
```

If there are debug-logs I can't find them (I have an unsupported HA install in docker, the Goodwe integration was installed using HACS)

spockz, 2022-06-11:
I have the same issue. Debug logging can be enabled using:

```yaml

logger:
  default: info
  logs:
    custom_components.sems: debug

```

When I do it says that when doing the query the authentication is expired.

ivo-toby, 2022-06-11:
Thanks, enabled debug, but no clue to where the logs are supposed to be.

sunib, 2022-06-11:
I see the same symptoms in my system, it has been working flawless for months but suddenly stopped 3 days ago! No time yet allocated to analyse it further.

The sems app does show data so the inverters are running as expected.

ivo-toby, 2022-06-11:
last update of the (ios) app was 1 month ago.. They possibly changed the auth-flow and updated the apps in advance.
I have created an NodeJs API client a while ago, i'll check if that still works

ivo-toby, 2022-06-11:
My (very old) API Client does not work anymore as well. Auth calls return 

``` 
{
  language: 'zh_CN',
  function: null,
  hasError: true,
  msg: 'ver is not fund',
  code: 'UnknownVer',
  data: null,
  components: {
    para: null,
    langVer: 143,
    timeSpan: 0,
    api: null,
    msgSocketAdr: null
  }
}
```

At the time I created the client they didn't have a portal so I used Charles Proxy to check the calls the iOS app made, this does not work anymore as they've added SSL-pinning. I guess the portal-api is next in line to be reverse engineered

Tonny56, 2022-06-12:
Same here. Stopped working yesterday evening.
Logger: custom_components.sems.sems_api
Source: custom_components/sems/sems_api.py:125
Integration: GoodWe SEMS API (documentation, issues)
First occurred: 11 juni 2022 21:25:40 (1 occurrences)
Last logged: 11 juni 2022 21:25:40

Unable to fetch data from SEMS.

spockz, 2022-06-12:
Following an example on the issues which uses `eu.` I changed the [power station url](https://github.com/TimSoethout/goodwe-sems-home-assistant/blob/916ebe7d04065fea066ac2ac4110b12ed3cbb56c/custom_components/sems/sems_api.py#L12) in my local install to `eu.` instead of `www.` and for now it seems to work.

spockz, 2022-06-12:
And... it seems this was fixed in https://github.com/TimSoethout/goodwe-sems-home-assistant/releases/tag/3.7.0-beta. :)

Tonny56, 2022-06-12:
@spockz Thank you, this has fixed it!

TimSoethout, 2022-06-12:
Thanks @spockz and @Tonny56 . Did 3.7.0 beta fix it? If so I'll promote it to a normal release.

micheldejongh, 2022-06-12:
@TimSoethout 3.7.0 Beta fixed it for me as well

RamonM77, 2022-06-12:
@TimSoethout  3.7.0 beta fixed the issue for me too

sunib, 2022-06-13:
Thanks for the help all, updating to 3.7.0 also solved it for me.

commit 04bb9b8a (2022-06-13): Release 3.7.0

Fixes #68