TimSoethout/goodwe-sems-home-assistant #75: API stopped working?

ruurdsanders, 2022-10-09:
Hello, I’ve been using this integration for about 1 1/2 months now and it’s been pretty stable. Until last Saturday at 0:08 however it completely stopped working and no more data flowed in. I tried reinstalling the interface with the right credentials but it does not provide any entities anymore. I also tried installing an older version but the same issue. What could be wrong here?

ruurdsanders, 2022-10-09:
This is what the logs say 
2022-10-09 11:52:58.704 ERROR (SyncWorker_7) [custom_components.sems.sems_api] Unable to fetch data from SEMS.
2022-10-09 11:52:58.726 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up sems platform for sensor

ruurdsanders, 2022-10-09:
The full logs:
Logger: homeassistant.components.sensor
Source: helpers/update_coordinator.py:178
Integration: Sensor (documentation, issues)
First occurred: 23:11:51 (1 occurrences)
Last logged: 23:11:51

Error while setting up sems platform for sensor
Traceback (most recent call last):
  File "/config/custom_components/sems/sensor.py", line 57, in async_update_data
    inverters = result["inverter"]
TypeError: 'NoneType' object is not subscriptable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
    self.data = await self._async_update_data()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 164, in _async_update_data
    return await self.update_method()
  File "/config/custom_components/sems/sensor.py", line 97, in async_update_data
    raise UpdateFailed(f"Error communicating with API: {err}")
homeassistant.helpers.update_coordinator.UpdateFailed: Error communicating with API: 'NoneType' object is not subscriptable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 281, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/sems/sensor.py", line 118, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_config_entry_first_refresh
    raise ex
homeassistant.exceptions.ConfigEntryNotReady: Error communicating with API: 'NoneType' object is not subscriptable

ruurdsanders, 2022-10-09:
Can someone please tell me WTH is going on?

TimSoethout, 2022-10-10:
Hi Ruud,

Not sure what the exact issue is. It seems that maybe the API result does not contain the "inverter". 

- Did you rename the inverter recently?
- Can you check that the inverter is visible in the app or on semsportal.com?

ruurdsanders, 2022-10-10:
The issue is that I didn't change anything. The inverter name is what was originally created by the integration. The inverter is also perfectly visible in the SEMS portal with the proper Power Station ID

ruurdsanders, 2022-10-10:
I also tried a new user but that didn't result in anything. The whole interface just keeps failing. The only thing I can think of is that an HA upgrade broke it but the last upgrade I did is 2 weeks ago. Also it wouldn't explain why all of a sudden it stops working?

ruurdsanders, 2022-10-10:
I also tried different versions of your interface @TimSoethout but the result is the same

ruurdsanders, 2022-10-11:
@TimSoethout any thoughts on what this could be?

TimSoethout, 2022-10-12:
No idea.

Maybe you can check the API output? 
See https://github.com/TimSoethout/goodwe-sems-home-assistant/issues/64#issuecomment-1048551841 for example using `httpie`.

The component does the same http calls.

ruurdsanders, 2022-10-12:
Sorry I have no clue what you mean?

notr1ch, 2022-11-15:
I also seem to be running into this.

```
2022-11-15 13:34:14.183 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Login Response: <Response [200]>
2022-11-15 13:34:14.183 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - API Token received: {'uid': 'xxx', 'timestamp': 1668515654085, 'token': 'xxx', 'client': 'ios', 'version': '', 'language': 'en', 'api': 'https://eu.semsportal.com/api/'}
2022-11-15 13:34:14.184 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Querying SEMS API (https://eu.semsportal.com/api//v2/PowerStation/GetMonitorDetailByPowerstationId) for power station id: xxx
2022-11-15 13:34:14.642 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] Query not successful (未将对象引用设置到对象的实例。), retrying with new token, 1 retries remaining
2022-11-15 13:34:14.642 DEBUG (SyncWorker_4) [custom_components.sems.sems_api] SEMS - Making Power Station Status API Call
2022-11-15 13:34:14.642 INFO (SyncWorker_4) [custom_components.sems.sems_api] SEMS - Maximum token fetch tries reached, aborting for now
2022-11-15 13:34:14.643 ERROR (SyncWorker_4) [custom_components.sems.sems_api] Unable to fetch data from SEMS. 
2022-11-15 13:34:14.651 DEBUG (MainThread) [custom_components.sems.sensor] Resulting result: None
```

robbeers76, 2023-02-04:
I have the same problem as Ruurd. Scraping from sems-portal stopped working after 2 months. I also cannot connect to semsportal.com website and my  inverters are off-line. I think I am blacklisted at the semsportal-server. Probably because I scraped every 60sec. Are there more people with this problem? Contact with Goodwe helpdesk did not help so far.

TimSoethout, 2023-02-04:
That's too bad @robbeers76 . I'm also scraping at the default rate of 60s, and am not banned. I have one only inverter though, maybe that's a difference?

Does it differ is you use www.semsportal.com or eu.semsportal.com, or other local ones?

robbeers76, 2023-02-04:
I can't connect to both sites with al my devices at home. a factory default
of my router/modem did not help either.
Outside my home I can connect en login to semsportal and notice that my
inverters are offline. The inverters are connected to my router but also
cannot connect to the semsportal server
That I cannot open the websites makes me believe that I am banned.

oakedcorn, 2023-02-07:
@robbeers76, I have exactly the same problem since end januari, semsportal cannot be accessed from within the home network (ISP IP address). It looks indeed like a banned IP address. I have 2 inverters updating at the default interval. I logged the issue with goodwe, but so far did not receive any response.

robbeers76, 2023-02-08:
Goodwe aswered. They don't banne anyone. With a ping-test I can reach the
website. So now I get with my problem to my internetprovider Youfone (via
KPN).
I also have 2 inverters and when I connect devices to the WIFI of my
neighbours they wil get online. So the problem must be an
internet-connection problem. But where?

oakedcorn, 2023-03-23:
I received feedback from my ISP, apparently the problem lies with twelve99.net and the ISP has no way to fix this because it's out of their control.

TimSoethout, 2023-03-23:
Too bad. :(
Also, my DNS can't resolve twelve99.net .

robbeers76, 2023-03-26:
Now I switches internetproviders (Budget via KPN) and now I have an other WAN-Adres. This was the solution and connection to the semsportal is working. I have two inverters and I have set the update time at 89sec and the other at 90sec so the update wil not be at the same time. I hope the connection won't stop working over 3 mnths.