TimSoethout/goodwe-sems-home-assistant #77: SEMS API show only two devices an 4 entities

matusrajnoha, 2022-10-13:
[image]

Please, how to resolve?

TimSoethout, 2022-10-13:
> [https://user-images.githubusercontent.com/109215057/195651401-9f4a1730-a35b-4339-a281-61a99bc9f9de.png][Snímka obrazovky 2022-10-13 o 18 19 54][https://user-images.githubusercontent.com/109215057/195651401-9f4a1730-a35b-4339-a281-61a99bc9f9de.png]
>
> Please, how to resolve?

What did you expect? Each inverter has 2 entities.

matusrajnoha, 2022-10-13:
I can see a lot of entities over a local connection. In the past I also had them through the SEMS API
[image]

TimSoethout, 2022-10-13:
> [https://user-images.githubusercontent.com/109215057/195694059-afb9abc4-f815-40e9-a270-1e3a59c83163.png][Snímka obrazovky 2022-10-13 o 21 39 45][https://user-images.githubusercontent.com/109215057/195694059-afb9abc4-f815-40e9-a270-1e3a59c83163.png]

The entities contain the values as attributes.

matusrajnoha, 2022-10-13:
The question was why I don't have them there when they were there.

TimSoethout, 2022-10-14:
They were there when using the local component if I understand you correctly. That is a different component which works differently. Maybe if you use that one again they will return?

This component is fetching the data via the cloud API. The other one, which is included in HA, uses the local network directly to the inverter.

Is this the issue?

matusrajnoha, 2022-10-14:
Yes, it's a problem. I didn't use the local component because of the floods. Entities and sensors were available. Now for some reason they don't. You state this, see photo.
[image]

TimSoethout, 2023-01-04:
Yes, the photo shows attributes, not separate entities. The screenshot is old so maybe the UI looks different now, but the attributes are still there. :)

closed 2023-01-04 by TimSoethout