TimSoethout/goodwe-sems-home-assistant #78: Integration is using deprecated `DEVICE_CLASS_*` constants

frenck, 2023-01-03:
This custom integration uses deprecated `DEVICE_CLASS_*` constants in its codebase.

The `DEVICE_CLASS_*` constants have been deprecated and replaced in Home Assistant Core 2021.12 (over a year ago). I would highly suggest updating/migrating this integration to the new enums.

For example, for the device classes supported by the `sensor` platform, there is now a `SensorDeviceClass` enum. So if a sensor previously used the `DEVICE_CLASS_ENERGY` constant, it should now use `SensorDeviceClass.ENERGY`. Other platforms (like `binary_sensor`, and `number`) provide similar enumerations for their supported device classes.

The migration thus only consists of replacing constants with an enumeration member and is, therefore, very low impact and should be fairly straightforward.

If I can help resolve any questions regarding this change or migration, feel free to ask or respond to this issue. I'm happy to help!

Kindest regards,

../Frenck

TimSoethout, 2023-01-04:
Thanks @frenck. I will definitely take this into account the next time I'm making changes to the code.
Feel free to submit a PR if you need this earlier.

closed 2024-03-11 by TimSoethout

frenck, 2024-03-11:
Thanks for fixing @TimSoethout 👍 

../Frenck

TimSoethout, 2024-03-11:
Main contribution is from @markruys. Thanks, @markruys .

PR #102 (2024-03-11): Use new constants
Fixes #78 and #96.