TimSoethout/goodwe-sems-home-assistant #79: No Entities

RemiKok63, 2023-01-04:
I tried to implement the sems portal integration, but it doesn't show any entities. Error is can't fetch data. I checked id and login they are right.

Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.sems.sems_api
Source: custom_components/sems/sems_api.py:123 
Integration: GoodWe SEMS API (documentation, issues) 
First occurred: 3 januari 2023 om 15:07:31 (2 occurrences) 
Last logged: 15:28:59

Unable to fetch data from SEMS.

RemiKok63, 2023-01-05:
Used the main account, problem solved. Seems not working with visitor account

Spyky, 2023-02-03:
Hello,
i'm new to homeassistant and want to add Goodwe sems. 1 only see 2 entities. The reame says you can add more.
In which file i have to add that?

TimSoethout, 2023-02-03:
Hi @Spyky , in your configuration.yaml. please see the home assistant documentation on template sensors.

Closing this since original issue of original poster was already solved.

closed 2023-02-03 by TimSoethout