TimSoethout/goodwe-sems-home-assistant #81: New user - is it true this HACS only exposes two sensors?

old-square-eyes, 2023-02-25:
I'm a bit of a noob so sorry if my termanology is off or I missed something. I was able to add HACS, and then SEMS to my Home Assistant Docker Setup and see only "Total Generation" and "Current Generation" from my Goodwe panels.

I would like to automate my GPU mining rig based on Nicehash (yes I know GPU mining is dead) to only mine when I am generating excess energy.

The easiest way to do that is to periodiacally monitor at a threshold when I'm pushing to and pulling from the grid.

How can I bring the grid import/export attribute in to process this info?

old-square-eyes, 2023-02-25:
Actually I see some import and export figures under Homekit. What are these? They don't match my Goodwe SEMS mobile app and are measured in kWh.

midadami, 2023-02-25:
SEMS works badly from time to time. It has several minutes delay when it works badly. It has months without problems but since yesterday its again failing.

old-square-eyes, 2023-02-25:
> SEMS works badly from time to time. It has several minutes delay when it works badly. It has months without problems but since yesterday its again failing.

Wait so I should be able to see more data ordinarily?

midadami, 2023-02-25:
> > SEMS works badly from time to time. It has several minutes delay when it works badly. It has months without problems but since yesterday its again failing.
> 
> Wait so I should be able to see more data ordinarily?

Under current generation you can see all the inverter attributes when it works correctly. SEMS not working correctly for me now.

TimSoethout, 2023-02-25:
What @midadami says indeed. In the readme (https://github.com/TimSoethout/goodwe-sems-home-assistant#screenies) you see a screenshot of some of the attributes. There are more when you scroll down at your own instance.
Hopefully there is an attribute there that you can use for your purpose.

old-square-eyes, 2023-02-25:
I'm pretty confused then. I see nothing like the screen shot. I see the below.

[image]

old-square-eyes, 2023-02-25:
Clicking into the graphs only gives me timelines of the existing metrics,...

[image]

TimSoethout, 2023-02-25:
> Clicking into the graphs only gives me timelines of the existing metrics,...
> 
> [image]

Click on `Attributes` in the lower right of the picture. :)

old-square-eyes, 2023-02-25:
I see it now. Not sure if I missed it before or if it wasn't there. Thanks! Not sure which one's export but I'll find it I'm sure.

old-square-eyes, 2023-02-25:
FYI for anyone else I used this to set up a sensor for the attribute values...

https://community.home-assistant.io/t/lovelace-ui-gauge-card-add-attribute-variable/311634/3

[image]

This is great!

closed 2023-02-25 by TimSoethout