TimSoethout/goodwe-sems-home-assistant #82: Ability to change server

luuks123, 2023-03-21:
Is it possible to use another server of Semsportal, us.semsportal.com for example ?

It always connects to eu.semsportal.com, but that isnt working in my case. I want to use us.semsportal.com because that one is working fine.

TimSoethout, 2023-03-21:
Easiest way is to change the hardcoded URL in the source code in your `custom_components` directory. 
The component does not support changing this via configuration currently.

luuks123, 2023-03-21:
I have tried that, but it didnt work. 

i changed the url and rebooted HA but it keeps connecting to eu.semsportal.com, is there something i'm doing wrong ?

TimSoethout, 2023-03-21:
That is what I would expect to work indeed. Does it show the eu URLs in the debug logs?

luuks123, 2023-03-21:
De debug log still connects to eu.semsportal.com,