TimSoethout/goodwe-sems-home-assistant #83: brand logo

Arrheniu2, 2023-03-28:
Hi, im not sure if is an issue but we cant see the brand sems logo in the integration, maybe we can add it to the home assistants brand logo repositore? 

not sure if we can pull that logo there to the domain "sems" or maybe we should host locally or add to this repositore

TimSoethout, 2023-03-28:
Great idea. Feel free to make a pull request with the required changes. :)

Arrheniu2, 2023-03-28:
i made my first github pull request here: 
https://github.com/home-assistant/brands/pull/4251

probably i messed up something but eventually we will get the logo in home assistant (i hope)

thanks for your work, tho.
regards

f00f, 2023-10-31:
I guess you should resubmit that PR with the new Goodwe brand logo which is now a red stylized letter 'W'.

TimSoethout, 2025-03-24:
Fixed for now. For a new logo a new issue/PR can be used. :)

closed 2025-03-24 by TimSoethout