TimSoethout/goodwe-sems-home-assistant #84: Optional data

Kotikoodari, 2023-04-01:
Thanks for the SEMS API!
I have just installed that and got it preliminary operational.

However, I wonder why it doesn't provide all the available data - such as daily energy production.
Current power and total Energy are provided.

I'm trying to follow instructions and add temperature and daily energy generation visible by adding "PV Temperature" and "PV energy day".
However, the instructions are not very clear for a new HA guy like me and I hope to get very clear instructions how to get the data available.

Instruction tells:

**Extra (optional) templates to easy access data as sensors**
Replace $NAME with your inverter entity id.

I have created a sensors.yaml file (in the same folder as configurations.yaml) with below text (with correct id):

```
- platform: template
  sensors:
    pv_temperature:
        value_template: '{{ states.sensor.inverter_$NAME.attributes.tempperature }}'
        unit_of_measurement: 'C'
        friendly_name: "PV Temperature"
    pv_eday:
        value_template: '{{ states.sensor.inverter_$NAME.attributes.eday }}'
        unit_of_measurement: 'kWh'
        friendly_name: "PV energy day"
    pv_etotal:
        value_template: '{{ states.sensor.inverter_$NAME.attributes.etotal }}'
        unit_of_measurement: 'kWh'
        friendly_name: "PV energy total"

```

And I have written a row in the configurations.yaml:

`sensor: !include sensors.yaml`

However, I cannot get this working...

When trying to add a sensor "PV energy day" it says that it is not in use and it doesn't provide value.

Can someone help to provide clear instructions... I'm not too familiar yet with the logic of the yaml files and templates...

Thank you!

sixtybi, 2023-07-26:
Hi, have you figured this out already? I just faced the same issues here. Not very clear how to fetch more data. I only have current power and generated total.
Please let me know how to proceed with this. Thanks!

patrickmoll83, 2023-08-02:
I have the dame issue, can someone help me?

f00f, 2023-10-31:
For me, this worked:

```yaml
template:
  unique_id: pv_gw10k_et
  sensor:
    - name: "PV Energy day"
      unique_id: pv_gw10k_et.pv_energy_day
      device_class: energy
      state: '{{ states.sensor.inverter_gw10k_et.attributes.eday }}'
      state_class: total
      unit_of_measurement: kWh
    - name: "Battery SoC"
      unique_id: pv_gw10k_et.pv_soc
      state: '{{ states.sensor.inverter_gw10k_et.attributes.soc }}'
      state_class: measurement
      unit_of_measurement: '%'
      icon: mdi:battery-20
    - name: "Battery SoH"
      unique_id: pv_gw10k_et.pv_soh
      state: '{{ states.sensor.inverter_gw10k_et.attributes.soh }}'
      state_class: measurement
      unit_of_measurement: '%'
      icon: mdi:battery-check-outline
```

ron84088, 2025-03-14:
> For me, this worked:
> 
> template:
>   unique_id: pv_gw10k_et
>   sensor:
>     - name: "PV Energy day"
>       unique_id: pv_gw10k_et.pv_energy_day
>       device_class: energy
>       state: '{{ states.sensor.inverter_gw10k_et.attributes.eday }}'
>       state_class: total
>       unit_of_measurement: kWh
>     - name: "Battery SoC"
>       unique_id: pv_gw10k_et.pv_soc
>       state: '{{ states.sensor.inverter_gw10k_et.attributes.soc }}'
>       state_class: measurement
>       unit_of_measurement: '%'
>       icon: mdi:battery-20
>     - name: "Battery SoH"
>       unique_id: pv_gw10k_et.pv_soh
>       state: '{{ states.sensor.inverter_gw10k_et.attributes.soh }}'
>       state_class: measurement
>       unit_of_measurement: '%'
>       icon: mdi:battery-check-outline

Hi @f00f,
yes, this is how it works.
Could you also figure out, how to get the 'More Batterys' attributes working, also?

More batterys
- pbattery: 52.48
vbattery: 262.4
ibattery: 0.2
battary_work_mode: 2
battstrings: 5
bms_status: 1
bms_temperature: 11
bms_discharge_i_max: 50
bms_charge_i_max: 50
bms_alarm_l: 0
bms_warning_l: 0
soc: 44
soh: 100
bmsbattstr: 5
batteryprotocol: 261
bms_alarm_h: 0
bms_warning_h: 0
bmssoftwareversion: 792
batteryhardwareversion: 0
batttotalcharge: 0
batttotaldisCharge: 0
batterysnmain: ''
bms_alarm: 0
bms_warning: 0
- pbattery: 78.84
vbattery: 262.8
ibattery: 0.3
battary_work_mode: 2
battstrings: 5
bms_status: 1
bms_temperature: 11
bms_discharge_i_max: 50
bms_charge_i_max: 50
bms_alarm_l: 0
bms_warning_l: 0
soc: 46
soh: 100
bmsbattstr: 5
batteryprotocol: 261
bms_alarm_h: 0
bms_warning_h: 0
bmssoftwareversion: 792
batteryhardwareversion: 0
batttotalcharge: 0
batttotaldisCharge: 0
batterysnmain: ''
bms_alarm: 0
bms_warning: 0

I have two batteries, so I could use some of those ;-)