TimSoethout/goodwe-sems-home-assistant #85: Component not starting - TypeError: 'NoneType' object is not subscriptable

justinmaiuto, 2023-04-15:
Since the 2023.4.x upgrade, the HA SEMS component will not start. The actual SEMS portal is working fine and receiving/logging data. 

```
Logger: homeassistant.components.sensor
Source: helpers/update_coordinator.py:222
Integration: Sensor (documentation, issues)
First occurred: 07:09:58 (1 occurrences)
Last logged: 07:09:58

Error while setting up sems platform for sensor
Traceback (most recent call last):
  File "/config/custom_components/sems/sensor.py", line 57, in async_update_data
    inverters = result["inverter"]
TypeError: 'NoneType' object is not subscriptable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 250, in _async_refresh
    self.data = await self._async_update_data()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 206, in _async_update_data
    return await self.update_method()
  File "/config/custom_components/sems/sensor.py", line 97, in async_update_data
    raise UpdateFailed(f"Error communicating with API: {err}")
homeassistant.helpers.update_coordinator.UpdateFailed: Error communicating with API: 'NoneType' object is not subscriptable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 304, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/sems/sensor.py", line 118, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 222, in async_config_entry_first_refresh
    raise ex
homeassistant.exceptions.ConfigEntryNotReady: Error communicating with API: 'NoneType' object is not subscriptable
```

justinmaiuto, 2023-04-15:
Seems it is the same issue as here?

https://github.com/TimSoethout/goodwe-sems-home-assistant/issues/75

TimSoethout, 2026-02-06:
Closing, please open if still relevant. :)