TimSoethout/goodwe-sems-home-assistant #87: turn off invertor via api

tabnul, 2023-05-27:
With dynamic contracts it make sense to turn off the invertor during negative price hours. (Automated of course ;) )
Sems portal allows you to turn off (and resume) the power delivery.
Would it be possible to add it to the integration ?

Didnt look at the api myself but will do it somewhere this week

tabnul, 2023-06-02:
I was not entirely correct. The semsportal App does allow you to turn off and turn on power delivery. It is somewhat unstable , have to try it multiple times before it actually acts.
The webui is not showing it.

However i found that the functions are available on the website and you can get the control box to show to adjust the following code (just remove the class=app and it will show up in the gui)
 

`   <div class="app" id="control_form">
        
        <div class="gdw-message-box-header">
            <button type="button" class="btn-primary gmbh-selected-tags" v-on:click="op_form_panel(true)">Parameter control</button>
            <button type="button" class="btn-primary" v-on:click="op_form_panel(false)">Start and stop control</button>
        </div>`

Turning it off is a POST request on:

MZ-CZ, 2024-01-03:
Hi tabnul, you posted here scripts for turning on/off of goodwe inverter. Did you try it for change of Power export limit value by this way? Idealy not for inverter but for datalloger before inverter (EZ logger/SEC1000S). Thank you very much. I tried it through DRED inputs on dattaloger and it don't work for me...

tabnul, 2024-01-04:
> Hi tabnul, you posted here scripts for turning on/off of goodwe inverter. Did you try it for change of Power export limit value by this way? Idealy not for inverter but for datalloger before [...]

Hi, i moved to lan adapter and access it kocally now (no sems) so dont have an answer.

cthulu, 2024-02-12:
Has anyone attempted to implement this using SEMS API? I might be able to give it a go at some point...
Also, what is the LAN adapter? Can you communicate with the inverter via ETH cable without ever going through SEMS API (which is not great from the looks of the API)

MZ-CZ, 2024-02-12:
SEMS API way is possible only for single inverter. Not for multiple inverter systems with EZlogger or SEC1000.  It is slow regulation with neccesary internet connection for its work.
LAN (or WiFi) adapter is used by this integration (I think UDP protocol). When you make path through internet (VPN?) to inverter from your local network with regulator, it will work as is.
DRED or RCR inputs of inverter are not usable at Goodwe for this job. They regulate complete power of inverter not only export to grid.
I will use RS485 way for multiple inverteres.

cthulu, 2024-02-12:
As I only have a single inverter this might still be an interesting option for me, without the need to purchase controller. Granted, better solution is a full control to regulate the output, not just shut it down completely...

cthulu, 2024-03-09:
@tabnul how did you actually get to that call in your browser? I can't seem to find it in my options in SEMS portal (I can execute it in the iOS app but that's harder to troubleshoot). I've started making changes in the SEMS plugin for HA to try to expose this as a switch device, just need to know the call.
I 've tried it with Postman but getting

```
{
    "code": -100,
    "msg": "Network failure, please try again"
}
```

when doing `POST https://www.semsportal.com/GopsApi/Post9?s=PowerStation/SaveRemoteControlInverter` with JSON payload as described above.

tabnul, 2024-03-09:
> @tabnul how did you actually get to that call in your browser? I can't seem to find it in my options in SEMS portal (I can execute it in the iOS app but that's harder to troubleshoot). I've started [...]

If you do this you will see a hidden control bar: 

```
However i found that the functions are available on the website and you can get the control box to show to adjust the following code (just remove the class=app and it will show up in the gui)
It will show a tab for start-stop control.

  <div class="app" id="control_form">
```

cthulu, 2024-03-10:
Hm that worked to show the form, but the Power Control there doesn't do
anything. I see the 200 response but nothing happens on the inverter side.

tabnul, 2024-03-10:
> Hm that worked to show the form, but the Power Control there doesn't do anything. I see the 200 response but nothing happens on the inverter side.
> […](#)

To be honest i also didnt see it working but i assumed it would work after trying a few times , just like when you do it in the SEMS app, it takes a few times in the app as well.

The next best thing would be to set a proxy on your phone and use that while shutting down your invertor via the app (and look at the traffic in the proxy)

The easiest is to just get a LAN adapter and use the local protocol to read out metrics (needs a firmware update to certain level on your invertor)
Downside is that all the adapters (both WIFI and LAN) are very insecure / hard to secure. Its a mess tbh .

cthulu, 2024-03-10:
I did have a rooted Android phone laying around, and I've installed PCAPdroid and were able to sniff the HTTPS traffic. It hits the same endpoint `api/PowerStation/SaveRemoteControlInverter` however it's using `Content-Type: application/x-www-form-urlencoded`

It looks like it also sends the whole blob containing all the config variables, not just the "changed" power state (not sure if that matters, will need some testing). From the comparison, it looks like 
```
InverterStatus=2   #off
InverterStatus=4   #on
```

Here is an example form POST:
```
InverterSN=<REDACTED>&InverterRemotingLastSetDate=03%2F10%2F2024%2020%3A33%3A13&SaftyCountrySettingMark=0&SaftyCountry=0&PVStartVoltageSettingMark=0&PVStartVoltage=0&ReConnectTimeSettingMark=0&ReConnectTime=0&GridVoltageSettingMark=0&GridVoltageMax=0&GridVoltageMin=0&GridFrequencySettingMark=0&GridFrequencyMax=0&GridFrequencyMin=0&ReducedLoadParamSettingMark=0&OverFrequencyReducedLoadPoint=0&OverFrequencyReducedLoadSlope=0&LessFrequencyReducedLoadPoint=0&LessFrequencyReducedLoadSlope=0&ReactivePowerSettingMark=0&ReactivePower=0&ActivePowerLimitSettingMark=0&ActivePowerLimit=0&PowerFactorSettingMark=0&PowerFactor=0&StandardReactivePowerCurveSettingMark=0&StandardReactivePowerCurve=0&ISOLimitSettingMark=0&ISOLimit=0&InverterStatusSettingMark=1&InverterStatus=2&ShadowScanSettingMark=0&ShadowScan=0&GridPowerLimitSettingMark=0&GridPowerLimit=0&SendIntervalMark=0&SendInterval=0
```

cthulu, 2024-03-11:
Created a PR for this https://github.com/TimSoethout/goodwe-sems-home-assistant/pull/103
It worked in my testing...

closed 2024-03-20 by TimSoethout